vloschiavo/powerwall2 #1: Unable to retrieve authentication token

spoonwzd, 2018-04-04:
I'm having issues getting an auth token from my Powerwall 2.

The password is definitely correct (as it works in the Powerwall setup wizard) but no matter what I try, I keep getting a login error saying my credentials are wrong.

```
HTTP/1.1 401 Unauthorized
Access-Control-Allow-Credentials: false
Access-Control-Allow-Headers: X-Requested-With, X-HTTP-Method-Override, Content-Type, Accept, Accept-Encoding, Authorization
Access-Control-Allow-Methods: GET, POST, PUT, DELETE, PATCH, HEAD, OPTIONS
Access-Control-Allow-Origin: *
Access-Control-Max-Age: 86400
Content-Type: application/json
X-Content-Type-Options: nosniff
Date: Wed, 04 Apr 2018 20:13:32 GMT
Content-Length: 66

{"code":401,"error":"Invalid Credentials","message":"Login Error"}
```

Any ideas?

vloschiavo, 2018-04-04:
Let's see if we can figure this out. Can you share your request?

spoonwzd, 2018-04-04:
Thanks for your response.

Here's my request (with my IP & Powewall password obscured):

```
./curl -s -i -X POST -H "Content-Type: application/json" -d '{"username":"jo
e@rob.com","password":"STxxxxxxxxxx","force_sm_off":false}' http://192.168.xxx.xxx/api/login/Basic
```

Have tried it both with "" & an email address for username.

vloschiavo, 2018-04-04:
I'm on Powerwall version 1.15.0 and Ubuntu 16.04 LTS with curl version
7.47.0
curl --version
curl 7.47.0 (x86_64-pc-linux-gnu) libcurl/7.47.0 GnuTLS/3.4.10 zlib/1.2.8
libidn/1.32 librtmp/2.3
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3
pop3s rtmp rtsp smb smbs smtp smtps telnet tftp
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB
SSL libz TLS-SRP UnixSockets

What powerwall, OS and curl versions are you running?

Ok.  I just cut and pasted your request.  (I changed the password and ip to
match my system).

I received:
HTTP/1.1 200 OK
Access-Control-Allow-Credentials: false
Access-Control-Allow-Headers: X-Requested-With, X-HTTP-Method-Override,
Content-Type, Accept, Accept-Encoding, Authorization
Access-Control-Allow-Methods: GET, POST, PUT, DELETE, PATCH, HEAD, OPTIONS
Access-Control-Allow-Origin: *
Access-Control-Max-Age: 86400
Set-Cookie:
AuthCookie=g-p9-IVrYde5f_HphCp_9rF-ULmdfMVk7dpY7_fffffffffffffWYi347i1eV2EsYjM6pMWsteleOafdetfgf==;
Path=/
Date: Wed, 04 Apr 2018 21:36:30 GMT
Content-Length: 202
Content-Type: text/plain; charset=utf-8

{"email":null,"firstname":"Tesla","lastname":"Energy","roles":["Provider_Engineer"],"token":"g-p9-IVrYde5f_HphCp_9rF-ULmdfMVk7dpY7_fMGuwFQs4PBClcWYi347i1eV2EsYjM6pMWsteleOnezxatbw==","provider":"Basic"}
(Not that it probably matters, but I changed the auth token for this
email...)

_V

spoonwzd, 2018-04-04:
I'm in Windows 10. Have tried 7.59.0 and 7.53.1 and am running 1.15.0 Powerwall firmware.

vloschiavo, 2018-04-04:
So based on the response you are seeing, you have the IP address correct.  Let's focus on the serial.
How many characters are in your serial number? It should be a T followed by 10 digits....right?  So the password is "ST"+10 digits.

Also, make sure you're NOT using the Serial number of the Powerwall and you ARE using the serial number of the Gateway.

BJReplay, 2018-04-04:
FWIW, this code is what the PW service is running

  Dim LoginRequest As New LoginRequest With {
                    .username = My.Settings.PWGatewayUsername,
                    .password = My.Settings.PWGatewayPassword,
                    .force_sm_off = False
                }
                Dim BodyPostData As String = JsonConvert.SerializeObject(LoginRequest).ToString
                Dim BodyByteStream As Byte() = Encoding.ASCII.GetBytes(BodyPostData)
                Dim request As WebRequest = WebRequest.Create(My.Settings.PWGatewayAddress & "/api/login/Basic")
                request.Method = "POST"
                request.ContentType = "application/json"
                request.ContentLength = BodyByteStream.Length
                Dim BodyStream As Stream = request.GetRequestStream()
                BodyStream.Write(BodyByteStream, 0, BodyByteStream.Length)
                BodyStream.Close()
                Dim response As HttpWebResponse = CType(request.GetResponse(), HttpWebResponse)
                Dim dataStream As Stream = response.GetResponseStream()
                Dim reader As StreamReader = New StreamReader(dataStream)
                Dim responseFromServer As String = reader.ReadToEnd()
                Dim LoginResult As LoginResult = JsonConvert.DeserializeObject(Of LoginResult)(responseFromServer)
                PWToken = LoginResult.token

spoonwzd, 2018-04-04:
The serial number is correct - I can copy / paste it into the Gateway web interface configuration wizard AND it works with when placed in the config of https://github.com/BJReplay/PowerwallService whcih I'm also running.

spoonwzd, 2018-04-04:
Defo ST+10 digits

vloschiavo, 2018-04-04:
@spoonwzd - can you double check the command line options for windows curl?  I'm not sure if they are different than linux curl.

vloschiavo, 2018-04-04:
The other problem could be quoting.
Single versus double quote behavior on Linux vs Windows.

spoonwzd, 2018-04-04:
Looks right to me...

```
 -s, --silent        Silent mode (don't output anything)
 -i, --include       Include protocol headers in the output (H/F)
 -X, --request COMMAND  Specify request command to use
 -H, --header LINE   Pass custom header LINE to server (H)
 -d, --data DATA     HTTP POST data (H)
```

spoonwzd, 2018-04-04:
I had a little mess with quotes.

The URL works with nothing or "<content>" but not '<content>'

I figured I shouldn't mess with the '{JSON}' as it already contains quotes.

...and the debug output shows the content-type to be correct, so I've not touched that.

Anything else you can think of that I've not tried?

vloschiavo, 2018-04-04:
Try this:
./curl -s -i -X POST -H "Content-Type: application/json" -d "{\"username\":\"jo
e@rob.com\",\"password\":\"STxxxxxxxxxx\",\"force_sm_off\":false}" http://192.168.xxx.xxx/api/login/Basic

spoonwzd, 2018-04-04:
@vloschiavo It took the command OK, but still failed with invalid credentials :(

vloschiavo, 2018-04-04:
Ok, now I'm stumped.    Here's what I did on Windows 10 with curl 7.59.0

I tried your original request with the single quotes for the -d and it failed (401 Unauthorized)
`./curl -s -i -X POST -H "Content-Type: application/json" -d '{"username":"jo
e@rob.com","password":"STxxxxxxxxxx","force_sm_off":false}' http://192.168.xxx.xxx/api/login/Basic`

Then I changed the single quotes to a double quote and escaped each double quote inside the -d section with a backslash and got a 200 response with a new token.

`./curl -s -i -X POST -H "Content-Type: application/json" -d "{\"username\":\"jo
e@rob.com\",\"password\":\"STxxxxxxxxxx\",\"force_sm_off\":false}" http://192.168.xxx.xxx/api/login/Basic`

spoonwzd, 2018-04-04:
Success! The escaped character version worked.

Nice one :) 👍 

Thanks for your help.

vloschiavo, 2018-04-04:
Glad to hear it!  I'll make a note in the documentation.

vloschiavo, 2018-04-04:
Solved: Windows character escaping.

closed 2018-04-04 by vloschiavo