vloschiavo/powerwall2 #2: Updates to your docs

endoplasmic, 2018-04-26:
Hey hey,

I have the same setup as you do (2xPW2 in California) and wanted to help you fill in some blanks!

My part number is: `1092170-03-G`

`grid_faults` - This is either when I killed power to the grid while the gateway was still on or when I tried to connect to the gateway directly and it made itself drop the connections to the other services (solar, grid, home were reporting 0)

```js
{  
      "timestamp":1524712627604,
      "alert_name":"PINV_a008_vfCheckRocof",
      "alert_is_fault":false,
      "decoded_alert":"[{\"name\":\"PINV_alertID\",\"value\":\"PINV_a008_vfCheckRocof\"},{\"name\":\"PINV_alertType\",\"value\":\"Warning\"}]",
      "alert_raw":576460752303423488,
      "git_hash":"d94a493fe3f9bb",
      "site_uid":"12345678-00-J--TG12345678",
      "ecu_type":"TEPINV",
      "ecu_package_part_number":"1081100-79-J",
      "ecu_package_serial_number":"XXXXXXX"
   }
```

Thanks for writing this all up, it's super useful!

vloschiavo, 2018-04-29:
Thanks @endoplasmic! I appreciate the additional data!   I'll add this in.

ajdonnison, 2018-05-14:
Great documentation.  I found that the grid alerts include more than just power outages - in this case it reported an overvoltage on the grid (a regular problem around here).  In Australia so the nominal voltage is 230V and the alert seems to trigger at 260V:
```
  {
    "timestamp": 1526045792722,
    "alert_name": "PINV_a005_vfCheckOverVoltage",
    "alert_is_fault": false,
    "decoded_alert": "[{\"name\":\"PINV_alertID\",\"value\":\"PINV_a005_vfCheckOverVoltage\"},{\"name\":\"PINV_alertType\",\"value\":\"Warning\"},{\"name\":\"PINV_a005_ov_amplitude\",\"value\":260,\"units\":\"Vrms\"}]",
    "alert_raw": 360292372531118100,
    "git_hash": "9885ef25ae814a",
    "site_uid": "1111111-01-G--T17J0000000",
    "ecu_type": "TEPINV",
    "ecu_package_part_number": "1081100-99-J",
    "ecu_package_serial_number": "TG1173060001C5"
  }
```

closed 2019-10-03 by vloschiavo