vloschiavo/powerwall2 #5: Configuration changes don't take affect

spoonwzd, 2018-06-04:
So I'm having a go at making my own script in Powershell. I'm just at the 'working the commands out stage' but it's been going well.

I am able to set the mode & reserve percentage, but the settings are not taking effect after calling /api/config/completed.

Can you offer any advice? Here's my progress:

Invoke-WebRequest -Method POST -ContentType 'application/json' -Body '{"username":"dave@internet.com","password":"ST17K0001234","force_sm_off":true}' -Uri 'http://192.168.7.7/api/login/Basic'

```
StatusCode        : 200
StatusDescription : OK
Content           : {"email":null,"firstname":"Tesla","lastname":"Energy","roles":["Provider_Engineer"],"token":"3NSFEB
                    OeOaSSvSinb98WhZsFh4MwG9fVk6-xPy2bwdam6oCvDeYi1FvTIFtV6EUCSuBQC9ZFLfCQgu1uAqG5Yw==","provider":"Bas
                    ic...
RawContent        : HTTP/1.1 200 OK
                    Access-Control-Allow-Credentials: true
                    Access-Control-Allow-Headers: X-Requested-With, X-HTTP-Method-Override, Content-Type, Accept,
                    Accept-Encoding, Authorization
                    Access-Control-Al...
Forms             : {}
Headers           : {[Access-Control-Allow-Credentials, true], [Access-Control-Allow-Headers, X-Requested-With,
                    X-HTTP-Method-Override, Content-Type, Accept, Accept-Encoding, Authorization],
                    [Access-Control-Allow-Methods, GET, POST, PUT, DELETE, PATCH, HEAD, OPTIONS],
                    [Access-Control-Allow-Origin, *]...}
Images            : {}
InputFields       : {}
Links             : {}
ParsedHtml        : System.__ComObject
RawContentLength  : 202
```

Invoke-WebRequest -Uri 'http://192.168.7.7/api/sitemaster/run'

```
StatusCode        : 202
StatusDescription : Accepted
Content           :
RawContent        : HTTP/1.1 202 Accepted
                    Access-Control-Allow-Credentials: true
                    Access-Control-Allow-Headers: X-Requested-With, X-HTTP-Method-Override, Content-Type, Accept,
                    Accept-Encoding, Authorization
                    Access-Cont...
Forms             : {}
Headers           : {[Access-Control-Allow-Credentials, true], [Access-Control-Allow-Headers, X-Requested-With,
                    X-HTTP-Method-Override, Content-Type, Accept, Accept-Encoding, Authorization],
                    [Access-Control-Allow-Methods, GET, POST, PUT, DELETE, PATCH, HEAD, OPTIONS],
                    [Access-Control-Allow-Origin, *]...}
Images            : {}
InputFields       : {}
Links             : {}
ParsedHtml        : System.__ComObject
RawContentLength  : 0
```

$token = '3NSFEBOeOaSSvSinb98WhZsFh4MwG9fVk6-xPy2bwdam6oCvDeYi1FvTIFtV6EUCSuBQC9ZFLfCQgu1uAqG5Yw=='
$link = 'http://192.168.7.7/api/operation'
$header = @{Authorization = 'Bearer '+$token}

```
Invoke-WebRequest -Method POST -Body '{"mode":"self_consumption", "backup_reserve_percent":100}' -Uri $link -headers $header

StatusCode        : 200
StatusDescription : OK
Content           : {"mode":"self_consumption","backup_reserve_percent":100}
RawContent        : HTTP/1.1 200 OK
                    Date: Mon, 04 Jun 2018 21:08:40 GMT
                    Access-Control-Allow-Credentials: true
                    Access-Control-Allow-Headers: X-Requested-With, X-HTTP-Method-Override, Content-Type, Accept,
                    Accept-Encod...
Headers           : {[Date, System.String[]], [Access-Control-Allow-Credentials, System.String[]],
                    [Access-Control-Allow-Headers, System.String[]], [Access-Control-Allow-Methods,
                    System.String[]]...}
Images            : {}
InputFields       : {}
Links             : {}
RawContentLength  : 56
RelationLink      : {}
```

Invoke-WebRequest -Uri http://192.168.7.7/api/config/completed -headers $header -Method GET

```
StatusCode        : 202
StatusDescription : Accepted
Content           :
RawContent        : HTTP/1.1 202 Accepted
                    Date: Mon, 04 Jun 2018 21:08:44 GMT
                    Access-Control-Allow-Credentials: true
                    Access-Control-Allow-Headers: X-Requested-With, X-HTTP-Method-Override, Content-Type, Accept,
                    Accept...
Headers           : {[Date, System.String[]], [Access-Control-Allow-Credentials, System.String[]],
                    [Access-Control-Allow-Headers, System.String[]], [Access-Control-Allow-Methods,
                    System.String[]]...}
Images            : {}
InputFields       : {}
Links             : {}
RawContentLength  : 0
RelationLink      : {}
```

So all looks good.... but it doesn't start charging from the grid!

Everything looks good to me:

Invoke-WebRequest -Uri $link -Headers $header

```
StatusCode        : 200
StatusDescription : OK
Content           : {"mode":"self_consumption","backup_reserve_percent":100}
RawContent        : HTTP/1.1 200 OK
                    Date: Mon, 04 Jun 2018 21:09:04 GMT
                    Access-Control-Allow-Credentials: true
                    Access-Control-Allow-Headers: X-Requested-With, X-HTTP-Method-Override, Content-Type, Accept,
                    Accept-Encod...
Headers           : {[Date, System.String[]], [Access-Control-Allow-Credentials, System.String[]],
                    [Access-Control-Allow-Headers, System.String[]], [Access-Control-Allow-Methods,
                    System.String[]]...}
Images            : {}
InputFields       : {}
Links             : {}
RawContentLength  : 56
RelationLink      : {}
```

Any ideas?

spoonwzd, 2018-06-04:
Figured it out. You have to set /api/sitemaster/run AFTER setting the operation and /api/config/completed.

This is now working:

```
$powerwallurl = 'http://192.168.7.7'
$login = Invoke-WebRequest -Method POST -ContentType 'application/json' -Body '{"username":"dave@internet.com","password":"ST17K0001234","force_sm_off":true}' -Uri $powerwallurl/api/login/Basic | ConvertFrom-JSON
$token = $login.token
$header = @{Authorization = 'Bearer '+$token}
Invoke-WebRequest -Method POST -ContentType 'application/json' -Body '{"mode":"self_consumption","backup_reserve_percent":5}' -Uri $powerwallurl/api/operation -headers $header
Invoke-WebRequest -Uri $powerwallurl/api/config/completed -headers $header
Invoke-WebRequest -Uri $powerwallurl/api/sitemaster/run
```

vloschiavo, 2018-06-04:
Glad to hear it!

closed 2018-06-05 by spoonwzd