vloschiavo/powerwall2 #6: Watts Of Usage allowed

Ghawken, 2018-06-10:
Hi
Thanks a lot for this API summary - it is very useful.

Can ask what is the lowest level of Watts that you see?

For example:
Night now and have solar Watts at 340 positive or so from the api
Battery also reporting 670 Watts despite not being sued 

Both positive numbers

The Tesla App - report both as off - will have a look at the JavaScript and see whether a low level is hard coded within (but compressed JavaScript is somewhat unhelpful)

What is everyone using for OFF; no Solar generation; and no Battery Power/Discharging?

Thanks

Glenn

BJReplay, 2018-06-10:
You almost certainly have the CTs incorrectly set up by your installer.

Battery generally reports 0, -10 or +10 when it is in standby.

Solar also generally reports 0 when the sun is down.

However, the cabling for the battery and the PV inverter must be right next to each other; when the battery is charging overnight, solar shows 15 to 20.  This is indicative of a CT measuring something in an adjacent cable (e.g. by induction).

I'd get Tesla support or your installer to look at it.

Ghawken, 2018-06-10:
Thanks for the advice

The installers were complete idiots to be frank and wouldn’t rule out anything.   Getting them back - hmm. no thanks they took two days to install and that is with me making three trips to hardware store to get stuff they didn’t have!

The only caveat to this is - within the app and the IP address/page both Solar and Battery are 0 kW - despite this low level Watts reported by API meters aggregate page.
Generation and consumption otherwise seem okay during day.
The wizard reports acceptable CT readings when I last check checked -  but will do so again when max output generation.
Not entirely sure a CT issue....

Glenn

closed 2018-06-12 by Ghawken