vloschiavo/powerwall2 #9: API changed for 1.20.0

karllmitchell, 2018-07-03:
Hi,

My powerwall upgraded to 1.20.0, and my code no longer runs.  Any ideas?

$ curl -s http://${powerwall_ip}/api/system_status/soe
<a href="https://192.168.2.14/api/system_status/soe">Moved Permanently</a>.

$  curl http://192.168.2.14/api/meters/aggregates
<a href="https://192.168.2.14/api/meters/aggregates">Moved Permanently</a>.

karllmitchell, 2018-07-03:
Actually, it looks like http was previously acceptable, but now https is required.

Mike-53, 2018-07-06:
I had the same problem since update of my Powerwall2 to 1.20.0 last week.
My python scripts to retrieve the data from the powerwall did not work any more. 

I also found that https is now required, but when I tested it via Firefox,
Firefox complained the connection (in my private home network) was not secure:
"powerwall uses an invalid security certificate. 
The certificate is not trusted because it is self-signed. 
Error code: SEC_ERROR_UNKNOWN_ISSUER".

So I had to grant a "security exception" and everything went fine, as before.

To verify this, have a try e.g. with the (Linux) command line (where "powerwall" is the hostname or IP address of the Powerwall): 
_wget https://powerwall/api/system_status/soe --no-check-certificate_
or 
_wget https://powerwall/api/meters/aggregates --no-check-certificate_

Good luck!

karllmitchell, 2018-07-06:
For now I’ve switched to using curl -sk for my bash Powerwall->Influx scripts, and using https: instead of http:.  Will see if there’s any way to authenticate at a later date.  Glad it was such an easy fix.

karllmitchell, 2018-07-07:
It looks like the change to 1.20.0 has affected login credentials too.  http://{$ip_addr}/api/login/Basic returns nothing.  Switching to using curl -k and https gives {"code":401,"error":"bad credentials","message":"Login Error"}.  When I try to log in via the wizard via a web browser, the old method also fails.

BJReplay, 2018-07-07:
I've been told (but not tried - I'm still on 1.15.3) that {"username":"installer","password":"???????","email":"email@email.com","force_sm_off":false} works to log in.  @karllmitchell can you try a login with an additional email and advise if that works?

karllmitchell, 2018-07-07:
Just did:

curl -k -H 'Content-Type: application/json' -X POST -d '{"username'"${username}"'", "password":"'"${serialno}"'", "email":"'"${email}"'", "force_sm_off":false}' https://${ip_addr}/api/login/Basic

I tried every combination of the following:

usernames tried: installer, Installer, customer, Customer
passwords tried: Tesla password, gateway serial number with an S in front, reset password from wizard
email tried: personal email address only (same as Tesla login)

In each case, it was the same result:

{"code":401,"error":"bad credentials","message":"Login Error"}

Let me know if you see anything wrong.

karllmitchell, 2018-07-07:
Specifically, setting username to installer worked.  However, email was not required.

spoonwzd, 2018-07-15:
v1.20.0 made its way onto my Powerwall today. It's been a battle, but I'm making progress.

In regards to dealing with the certificate with Windows, there's a few steps:

1. Edit your local security policy and navigate to Computer Configuration / Windows Settings / Security Settings / Public Key Policies / Certificate Path Validation Settings. On the 'Stores' tab, tick both 'Per user certificate stores' and ensure the 'Root certificate stores' radio button is selected as 'Third-Party Root CAs'

2. Import the Certificate from your Powerwall into your Trusted Root Certification Authorities local certificate store

3. Edit your hosts file in Windows\System32\Drivers\Etc and add the host alias 'powerwall' for your Powerwall's IP address; ie:
192.168.1.2 powerwall

Now when accessing your Powerwall by https://powerwall you don't get any certificate errors and you both trust the certificate and are using a valid common name. The subject alternative names 'teg' and 'powerpack' are also listed in the certificate and could be used instead.

Still not figured out why Invoke-WebRequest in Powershell isn't playing ball...

spoonwzd, 2018-07-15:
@karllmitchell I still get 401 Bad Credentials. Have tried it with no email defined at all, email defined as "" but nothing seems to work.

I was able to log in initially using the serial, but then it gave me a new password. I can log into the WebUI as installer using the new password.

curl -s -i -X POST -H "Content-Type: application/json" -d '{"username":"installer","password":"4d#SDwq8","force_sm_off":true}' https://powerwall/api/login/Basic

Anyone have any ideas? @BJReplay @vloschiavo

AmateurCritic, 2018-07-18:
I'm running Gateway version 1.20 and have just got a raspberry pi running stretch, to pull the State of Charge, (soe) and write it to a file.

$ wget -q -O soe --no-check-certificate https://192.168.0.6/api/system_status/soe
Quiet operation, ignore certificate and (re)write to file "soe"

$ cat soe
{"percentage":39.1497885599822}

frownbreaker, 2019-04-18:
Thanks, AmateurCritic  

I used a Node-Red command based on your suggestion

wget -q -O /root/.node-red/soe --no-check-certificate https://192.168.0.209/api/system_status/soe

Then I read the file after 2 seconds, checking the exit code of the wget

enginehead-tes, 2019-08-18:
Hi,I'm struggling here. my powerwall is running 0.3.11.2 but when I use curl command below:
curl -k -s -i -X POST -H -v "Content-Type: application/json" -d '{"username”:”installer”,”password”:”*********”,"email”:”myemail@mydom.com",”force_sm_off":false}' https://192.168.x.xx/api/login/Basic
I get {"code":401,"error":"bad credentials","message":"Login Error"}
I've tried the gateway sticker password, the password created by the web interface
no user and "installer" as user. I have tried the command from my mac and linux (raspbian) all with the same results.

BJReplay, 2019-08-18:
Try with empty string or null for email

closed 2019-10-03 by vloschiavo