vloschiavo/powerwall2 #10: version 1.21

icepicknz, 2018-07-27:
Seems that I can't set backup mode anymore

Barrys-MacBook-Pro:~ barrymurphy$ curl -sk --header "Authorization: Bearer TjrnGcgW9qE4NUE69l6ed6GOz01N7nbAoB_CEZbLbyglSpqUr2VxcTd-iqlaby2YzwJw7TZ4EDkCTRZpzfh9zA==" -X POST -d '{"mode":"backup","backup_reserve_percent":15}' http://192.168.33.9:81/api/operation
{"code":500,"error":"RealMode must be either self_consumption or backup","message":"Unable to set operational mode"}Barrys-MacBook-Pro:~ barrymurphy$

icepicknz, 2018-07-28:
seems sending real_mode backup fixed this issue

spoonwzd, 2018-08-02:
Confirmed; real_mode resolves the 1.21 issue

Thanks @icepicknz ;)

simwood-simon, 2018-09-06:
Real_mode works for `backup` or `self_consumption` but not `autonomous`. It only seems possible to set that through the app now. This is the case regardless which mode the unit is currently in.

Anyone achieved a workaround?

`{"code":500,"error":"RealMode must be either self_consumption or backup","message":"Unable to set operational mode"}`

closed 2019-10-03 by vloschiavo