vloschiavo/powerwall2 #12: /api/system/update/status requires authentication

pgalbavy, 2018-12-08:
On my freshly installed PW2s and non-backup gateway in the UK, f/w 1.25.0 the endpoint /api/system/update/status requires authentication. Returns a 403 without.

Just playing with various manual GETs right now, but this seemed to be the only one that needed it in the top half of the list.

kylerove, 2019-04-13:
Same here. Tried `curl -k -s -i -X POST -H "Content-Type: application/json" -d '{"username":"installer","password":"SXXXXXXXXXX","email":"email@email.com","force_sm_off":false}' https://192.168.xxx.xxx/api/login/Basic` with correct ip address and gateway serial number and get a 401 bad credentials login error. Anyone have luck?

kylerove, 2019-04-14:
After digging, the answer was right on Tesla's website:
https://www.tesla.com/support/energy/powerwall/own/monitoring-from-home-network

username=customer
email=your Tesla account email address
password=last 5 digits of your gateway serial number

You can then reset/choose your own customer password, if you want to make it stronger.

closed 2020-04-11 by vloschiavo