vloschiavo/powerwall2 #15: Authentication on 1.37.1

estebanpapp, 2019-06-20:
Seems that authentication has changed again, getting:

`{"code":401,"error":"bad credentials","message":"Login Error"}`

This is the curl command:

`curl -s -i -k -X POST -H "Content-Type: application/json" -d '{"username":"":"password":"ST123456789","force_sm_off":false}' https://192.168.XXX.XXX/api/login/Basic`

I tried several other combinations (including the ones reported in other issues) to no avail. Anybody knows?

frood88, 2019-06-20:
I had to reset my password via the gateway’s web interface by entering the serial number.

It gives you a new string of characters as a password to use instead.

I’ve had to do it a few times now, not sure what triggered it specifically, but my theory is that it might have been me mucking around between force_sm_off=true and false to see if I could avoid stopping the powerwall to retrieve the current reserve value, and so perhaps any login error increments an internal bad password counter.

piersdd, 2019-06-21:
I have found `"force_sm_off": true` resolves this.

estebanpapp, 2019-06-21:
ok, after trying a hundred options, this worked:
```
curl -s -i -k -X POST -H "Content-Type: application/json" -d '{"username":"customer","password":"XXXXX","force_sm_off":false}' https://192.168.0.14/api/login/Basic
```
Being XXXXX the password I set in the UI
Got a response like:
```
HTTP/2 200 
set-cookie: AuthCookie=<COOKIE>; Path=/
set-cookie: UserRecord=<USERRECORD>; Path=/
content-type: text/plain; charset=utf-8
content-length: 193
date: Fri, 21 Jun 2019 02:43:41 GMT

{"email":"","firstname":"Tesla","lastname":"Energy","roles":["Home_Owner"],"token":"<TOKEN>","provider":"Basic"}
```

estebanpapp, 2019-06-21:
Actually, that token doesnt let me change the mode:
```
{"code":403,"error":"Unable to POST to resource","message":"User does not have adequate access rights"}
```

estebanpapp, 2019-06-21:
This worked:
1) Go to the gateway web page
2) Select installer
3) Select "Forgot password"
4) enter the Gateway serial number (complete) format like: STGXXXXXXXXXXXX
5) this will generate a new installer password format like XXXXXXXXXX
6) use that password:
```
curl -s -i -k -X POST -H "Content-Type: application/json" -d '{"username":"installer","password":"XXXXXXXXXX","force_sm_off":false}' https://192.168.XXX.XXX/api/login/Basic
```
The mode is no longer "mode" but "real_mode", getting it:
``` 
curl -s -i -k -H "Authorization: Bearer <TOKEN>" https://192.168.XXX.XXX/api/operation
```
returns:
```
HTTP/2 200 
content-type: text/plain; charset=utf-8
content-length: 59
date: Fri, 21 Jun 2019 03:06:30 GMT

{"real_mode":"self_consumption","backup_reserve_percent":5}
```
Changing the mode:
```
curl -s -i -k -X POST -H "Authorization: Bearer <TOKEN>" -d '{"real_mode":"self_consumption","backup_reserve_percent":24.6}' https://192.168.XXX.XXX/api/operation
```
And the config/completed command now needs the authorization token:
```
curl -s -i -k -H "Authorization: Bearer <TOKEN>" https://192.168.XXX.XXX/api/config/completed
```

closed 2019-06-21 by estebanpapp