vloschiavo/powerwall2 #18: Change the Powerwall mode and Reserve Percentage

HeraldCoupe, 2019-07-19:
I have a Powerwall 2 and backup gateway 2, which was installed only a couple of weeks ago.
I'm using Windows to try to change the Powerwall mode and reserve percentage but I'm stuck.

I've managed to get the authorization token successfully using:
curl -s -i -X POST -H "Content-Type: application/json" -d "{\"username\":\"customer\",\"password\":\"XXXXX\",\"force_sm_off\":false}" https://powerwall/api/login/Basic
(where XXXX is the last 5 characters of the gateway password)

I'm using this command to try to change the powerwall mode:
curl --header "Authorization: Bearer wWdvk08XHC385hSXixz0E7qhidU1yDG7teELKPR4M3gJFuIytweh_sanXZtS9ZPk6wD6j2f1CyhDF65S7MFD-Q==" -X POST -d "{\"mode\":\"backup\",\"backup_reserve_percent\":24.6}" https://powerwall/api/operation

But the response is:
{"code":403,"error":"Unable to POST to resource","message":"User does not have adequate access rights"}

I suspect my syntax is incorrect but I can't see what the problem is.
Can anyone help please?

BJReplay, 2019-07-19:
You need to be logged in as installer to set percentages and modes.

Try logging in as installer.

e123149, 2019-07-19:
Why is there an auxiliary control port on the gateway? Can we leverage a third party control device like an RTAC to control the PW?

HeraldCoupe, 2019-07-20:
So do I need to get an authorisation token using installer rather than customer? I presume I need my installers password then?

BJReplay, 2019-07-20:
Yes, and if they haven't provided it, you can reset the installer
password.  Can't recall off the top of my head exactly how, but involves
flicking the switch on the side of the powerwall

dwleckie, 2019-07-23:
Hello - new non programmer user here.  I need to find a method or script to automate an automatic or timed switch between powerwall modes.  Existing modes are basically worthless on there own to me but if I could switch to advanced time cost saving mode at the start of my peak demand period and switch back to backup only mode at the end of my peak demand period - that would make my system work the way I need it to.  Your thread shows the most of the elements (over my head) needed to accomplish the task.  Any thoughts?

bridiver, 2019-10-05:
@dwleckie I have been doing something similar and it has caused some problems. Apparently even changes made directly to the gateway API still have to go through Tesla's servers and sometimes it would get "stuck". Everything looked correct on the mobile app and the gateway UI, but when I called Tesla they said that their internal systems showed that it was still stuck in the previous mode. They told me it gets "confused" if you change modes frequently and after a long discussion I finally got them to file a bug for the operation of the powerwall being out of sync with the UI. For now I've settled on just leaving it at cost savings 50% which seems to prevent it from unnecessarily drawing on the powerwall during off peak times, but that won't work during the summer when I need more battery power during peak hours. I also made a feature request for a "dumb" schedule since they completely over-engineered cost savings mode and I encourage you to do the same. Apparently they have had a lot of requests from people in Phoenix (like me) because APS charges based on peak demand. If I use 4kW of peak power for one second of one day during the month I get charged almost $100, but my off-peak power is dirt cheap.

dwleckie, 2019-10-05:
I have been working with the Senior Project Manager for Powerwall at Tesla - Lindsay Stone.   She has been promising corrections and bug fixes since it was installed in June.   The dumb mode is exactly what Tesla will not do and is exactly what Tesla sold me - and that is exactly what I need.   Sounds about time for a class action law suit.  I have many others in various forums that are having the same or similar issues.

bridiver, 2019-10-05:
@dwleckie I'm also very upset that the powerwall is dependent on Tesla's servers. That's not a viable model for a product designed to last 10+ years. Tesla could go out of business, stop supporting old versions of the gateway, their servers can go down, etc... I have an issue filed here https://github.com/vloschiavo/powerwall2/issues/22 and I'm going to watch the communications back and forth between the gateway and tesla's servers using a mitm proxy to see if there are additional API calls that need to be made to avoid Tesla's servers

bridiver, 2019-10-05:
a less desirable option that wouldn't have the same problem is to disable the powerwall during off peak times, but then it won't kick in if you lose power

frood88, 2019-10-05:
@dwleckie @bridiver This is should be achievable without involving Tesla’s servers or the app as I am doing at the moment for other functions.

In general, I leave my Powerwall in self-powered mode the whole time and use a PowerShell scheduled task script to set the backup reserve percentage higher to force charging during off peak in winter based on time of day for load-shifting (as we have morning and evening peak and multiple shoulder periods, which the app doesn’t support), and also dynamically based on other API calls to weather and irradiance forecast services.

It involves sending an API config complete request followed by a stop and a run request to soft-restart the Powerwall - all takes a few seconds, and is all done between my server and the Powerwall locally - no Tesla server involvement, and takes effect immediately.
Happy to provide the critical elements of my script when I’m home after the weekend, if needed.

dwleckie, 2019-10-06:
You sound like the man I have been looking for!!!!   I would love to get that setup on my system.

dwleckie, 2019-10-06:
I have been setting my reserve power to 100% every night at 8:02pm and leaving it set there until the following morning around 10:00am when powerwall is completely recharged.   I don’t want to use powerwall outside of APS peak periods because we buy off peak at about $0.05 and sell to APS at $0.129 per kWh.  It’s manual but the only method that works for me at the moment.

bridiver, 2019-10-06:
@frood88 I had no idea that Tesla's servers were involved until I got into that state where everything showed the setting I thought should be active (app and backup gateway UI), but it wasn't behaving correctly. I have been issuing the config complete command, but that wasn't sufficient so maybe the soft reset is what was missing to actually put the mode into effect without waiting for whatever happens on Tesla's servers to activate it.

frood88, 2019-10-08:
@dwleckie @bridiver Just following up from a few days ago as promised.

As for most people, there is a whole lot of logic surrounding the way I use the API requests to decide what to set things to, when, checking whether a step has been successful, etc, but the critical components for the function I mentioned are:

1. Get token via: https://$PowerwallIPAddress/api/login/Basic
2. Set the backup reserve via: https://$PowerwallIPAddress/api/operation
 - with JSON as the body for: @{"real_mode"="self_consumption";"backup_reserve_percent"=$NewBackupReservePercent}
3. Commit the config via: https://$PowerwallIPAddress/api/config/completed
4. Cycle the Powerwall to make the changes take effect immediately by:
 - 4a. Get a new token (same as step 1) via: https://$PowerwallIPAddress/api/login/Basic
 - 4b. Resume the Powerwall via: https://$PowerwallIPAddress/api/sitemaster/run

Note: $PowerwallIPAddress and $NewBackupReservePercent are just PowerShell variables I use in my script - change the values for yourself as required

So essentially, during winter the battery is mostly depleted by 8-9PM due to ducted electric heating running during the evening peak tariff. So I set the backup reserve percent higher at about 3AM so that the Powerwall charges up to a certain percentage using off-peak power, and then 2 minutes before morning peak tariff starts, I set the reserve percentage back down to my default value.
I then also repeat the process in the early-mid-afternoon to ensure a certain percentage is reserved for the evening peak tariff.

Let me know if any questions!

bridiver, 2019-10-08:
@frood88 I had automation setup to verify the changes and alerts if they weren't set correctly at the correct times, but that didn't help me because the home gateway api was reporting the correct settings even though it wasn't actually in the correct mode. There can be a discrepancy between what is reported by the home gateway and the actual settings that are in force. Tesla support was able to see the actual mode in their internal systems and I feel like there must be some missing api calls that need to be discovered/documented.

enginehead-tes, 2019-10-08:
Hi if anyone is interested I've been using a Raspberry pi connected to the same network to control my Powerwall 2 for several weeks now. I'm based in the UK and have a electricity supplier who gives 30 minute pricing 24 hours ahead of time. I written a php script which calls the curl commands documented here to change the modes instantly, not via the tesla servers, to charge or discharge when prices are low/high respectively. I created a basic web interface on the pi to select the 30 slots. I'm happy to share my code and a guide FOC.
I'm using Mysql, Apache, PHP and Cur all installed on the P.l

bridiver, 2019-10-09:
I want to be very clear again that you may not think it is going through Tesla's servers because you are making direct API calls to the home gateway (as I am), but if you just change the mode and call `/config/completed` it still goes through Tesla's servers before the mode is actually changed. The mode displayed by the gateway UI is not always the mode that the powerwalls are actually in. I verified this on a call with Tesla where the gateway UI showed the mode I set it to as well as the mobile app, but it was actually still running in the previous mode and that was only visible through their internal systems. I was completely floored when Tesla support told me that even when making direct API calls to the gateway it still didn't actually take effect until it was updated by Tesla's servers. I'm hoping the additional steps of cycling the site manager will overcome that issue and force the change to apply without involving Tesla's servers, but I haven't had a chance to try it yet. Having any kind of dependency on Tesla's servers is completely unacceptable imo so hopefully @frood88's additional steps will force the change.

burble61, 2020-03-31:
> Hi if anyone is interested I've been using a Raspberry pi connected to the same network to control my Powerwall 2 for several weeks now. I'm based in the UK and have a electricity supplier who gives 30 minute pricing 24 hours ahead of time. I written a php script which calls the curl commands documented here to change the modes instantly, not via the tesla servers, to charge or discharge when prices are low/high respectively. I created a basic web GUI
> [image]
> on the pi to select the 30 slots. I'm happy to share my code and a guide FOC.
> I'm using Mysql, Apache, PHP and Curl all installed on the Pl

@enginehead-tes, would be very interested in seeing that.  Powerwall  installed late last year, took same supplier a while to get the Smart Meter installed and running (some circumstances outside their control). Mainly concentrated on logging activities up until now, but its become clear that the Tesla app needs augmenting to extract maximum value from the kit.

Happy to collaborate on producing documentation if of any value.

dwleckie, 2020-04-03:
THAT LLOKS ECELLENT!  Excited to see someone working on this.  Please share as much detail as possible (I am Raspberry Pi clueless however!)

 

 <https://user-images.githubusercontent.com/54237856/66440550-44c89480-ea2b-11e9-971e-e1dcc442b488.png> Screenshot 2019-10-09 at 00 21 44

bridiver, 2020-04-03:
@burble61 if you want to ensure the mode is changed without waiting for Tesla's servers you must cycle the powerwall state to stopped and back again. I'm not sure if you are already doing that or not, but as I mentioned https://github.com/vloschiavo/powerwall2/issues/18#issuecomment-539783368 just because you're hitting the powerwall api directly (as I was) does not mean the mode is always changed instantly and there are cases where the powerwall UI will "lie" to you. It will display the mode you expect, but it will not actually be in that mode until it gets cycled of and on. When you make changes through the mobile app it does cycle the sitemaster which is why you'll see the powerwall temporarily stop charging or discharging after you make a change.

bridiver, 2020-04-03:
and it seemed to work correctly for me most of the time without cycling the sitemaster right up to the point where mine got stuck because of an inconsistency between Tesla's servers and the powerwall. If you change the mode without cycling the powerwall, you are still dependent on Tesla's servers to commit the mode change and during that time the powerwall will show the mode you changed it to, not the mode it is actually in.

hdurdle, 2020-05-03:
@enginehead-tes  I'd be interested in seeing your code for updating the Powerwall based on your provider's rates.  I've written code to pull the next 24 hours of 30 minute periods from Octopus, and then output the period with the cheapest average electricity and the most expensive. I'm currently manually setting the Powerwall to charge in that cheapest period, and discharge in the peak.  Next step is to talk to the Gateway - but would rather not reinvent the wheel if you've already done it!

burble61, 2020-05-03:
> @burble61 if you want to ensure the mode is changed without waiting for Tesla's servers you must cycle the powerwall state to stopped and back again. I'm not sure if you are already doing that or not, but as I mentioned [#18 (comment)]..............

Understood, thankyou @bridiver ......  have manually operated the steps you highlighted, and verified Powerwall mode of operation indirectly via a separate energy monitoring system....   Next step is to script the changes and add a form of test harness to carry out that verification automatically over a larger number of the standard->cheap (and back) tariff transitions.

frownbreaker, 2020-06-05:
> Hi if anyone is interested I've been using a Raspberry pi connected to the same network to control my Powerwall 2 for several weeks now. I'm based in the UK and have a electricity supplier who gives [...]

I'd be interested just moved to Agile Octopus today. Currently, use the Time of day.
Want to track low/negative prices and then fill the 2 Powerwalls and the Car  when this happens
Any info you have welcome happy to post it on Gitlab and give you the credit under any open source licence you like Eg. GPL

Will update maintain as needed.

chrisduk112, 2020-06-07:
I’m also on octopus agile and looking to have powerwall go on backup only mode when octopus energy is on or below 0 pricing and change back to self powered when it above 0.01. 

^^ and tweak this in winter months to charge when under 5p

chrisduk112, 2020-06-08:
> Hi if anyone is interested I've been using a Raspberry pi connected to the same network to control my Powerwall 2 for several weeks now. I'm based in the UK and have a electricity supplier who gives [...]

Did you manage to document this process?

fkhera, 2020-10-27:
Anyone have any work around ideas 1.50.1 they locked down installer mode by requiring power toggling event

spoonwzd, 2020-10-27:
> Anyone have any work around ideas 1.50.1 they locked down installer mode by requiring power toggling event

Take a look at this thread [(https://github.com/vloschiavo/powerwall2/issues/34)]

closed 2021-02-21 by vloschiavo

greyghoster, 2023-08-28:
> 

I am having trouble replicating your success. Is it still working for you?