vloschiavo/powerwall2 #20: Missing Storm Watch setting and status?

gpeszek, 2019-08-17:
Been digging for a bit and can't seem to find where the Storm Watch setting and status is.  Looking mostly for the latter: how to tell if it is enabled and if it is activated.  Anyone have thoughts?

gpeszek, 2019-09-03:
Wondering if it is grid_services_active from api/system_status but since I haven't had a storm yet it's impossible for me to verify.  Anyone else know?

MikeBishop, 2019-12-30:
Also interested; I'd like to be able to trigger other things when Storm Watch activates.

HeraldCoupe, 2020-02-09:
As Storm Watch is actine now in the UK, has anyone found where the data is in the API?

MikeBishop, 2020-05-25:
It's in the cloud API, but not local that anyone has found so far. I just query the cloud endpoint every 30 minutes. Storm Watch is intended to come on hours in advance, so a 30-minute latency in detection doesn't hurt my scenarios.

closed 2021-02-21 by vloschiavo