vloschiavo/powerwall2 #22: mode changes don't always commit

bridiver, 2019-10-05:
I've been having issues where I make changes to the backup percentage and they show on the mobile app and the backup gateway UI, but they haven't actually been applied. I called Tesla to find out what was going on and they said that even if you make a change on the backup gateway, it still goes to Tesla's servers before it actually gets applied on the powerwall. That seemed a bit odd to me and I wasn't sure if I should believe a level 1 tech support rep on an issue like that, but several reps verified that there's no way for me to see the actual setting because the only way to view it is through their internal systems. This happens even when I call `/api/config/completed` after making the change. Is there possibly another API call that is missing from the list and is called by Tesla's servers to activate the change? I'm setting up a mitm proxy to try to capture what is happening

HeraldCoupe, 2019-10-08:
I've noticed this too - I think the answer you got from Tesla is erm... incorrect.
After changing the mode, I use /api/sitemaster/run to 'restart' the Powerwall which seems to kick it back into action.

bridiver, 2019-10-18:
@HeraldCoupe I know the answer they gave me is correct because the mobile app and the gateway UI both showed the mode I set, but that's not the mode it was actually in. They could see the correct mode in their internal systems and it was behaving according to that mode. The restart may bypass that, I still haven't had a chance to check yet

HeraldCoupe, 2019-10-18:
Thanks. This makes sense actually as last night my powerwall appeared to go into self-powered mode (initiated from my script) and yet it didn't start discharging but continued to draw from the grid.
I had to do another 'restart' to get it working.

bridiver, 2019-10-22:
I setup a mitm proxy to see what is happening, but haven't had a chance to look at it yet

closed 2021-02-21 by vloschiavo