vloschiavo/powerwall2 #28: What happened to Teg 1.45.1

fkhera, 2020-03-19:
Hello:

It seems as of update 1.45 TEG is not reachable on REST, did this happen to you guys?

spoonwzd, 2020-03-19:
Nooo don't tell me that :(

I'm still on 1.44.2

BJReplay, 2020-03-19:
I'd seen reports of that over at TMC.  Commentary suggested that the
gateway picked up a new IP where it was allocated a reserved DHCP lease,
and that it was suspected it had picked up a new MAC address.  Might be
worth checking your router to see if it has picked up a new IP.

1.45.1 is out, and reportedly fixed that problem.

Don't quote if if I haven't got that entirely right - quoting from memory.

So if you can find your gateway, perhaps try getting it to check for an
update.

I skipped from 1.43.1 to 1.45.1 without a problem.

fkhera, 2020-03-19:
I think its actually my IP changed.

spoonwzd, 2020-03-19:
Phew!

frood88, 2020-03-19:
Specifically, the last two or so version updates have resulted in the TEG’s MAC address changing, which automatically results in an IP change when DHCP in use, and especially so when an IP reservation is mapped to the MAC address.

BJReplay, 2020-03-19:
To the extent that this is a valid data point, I didn't see a change in IP
address jumping from 1.44.x to 1.45.1 - which might mean 1.45.1 incudes the
fix for this issue.

piersdd, 2020-03-20:
I can confirm, on 1.45.1, I have a new MAC address and IP address, AND the old/original statically mapped IP address still works.

closed 2020-04-11 by vloschiavo