vloschiavo/powerwall2 #33: Has anyone managed to get the PW2 to force discharge on command?

rolandh, 2020-10-07:
Is there anyway to force the PW2 to discharge on command regardless of household draw/usage?

This is a requirement for being able to rate surf. I get billed in 30 min intervals with peaks/troughs through out the day. Our electricity price can go negative at times and very high at others, being able to control charge/discharge would be amazing.

spoonwzd, 2020-10-07:
You can do this just by setting the reserve amount to different values - if the reserve % is set higher than the current stored amount the Powerwall will pull from the grid to charge to the defined % level. If the reserve % is set lower than the current stored amount the Powerwall will discharge.

daveoxley, 2020-12-10:
Setting the reserve amount will allow the discharge to match the current household demand. Is there anyway to force the powerwall to discharge to the grid?

vloschiavo, 2021-02-21:
I don't think this is possible.

closed 2021-02-21 by vloschiavo