vloschiavo/powerwall2 #34: API login/Basic fails on 1.50.1 for installer

rstad944, 2020-10-22:
For the last 16 months I have been able to use /api/login/Basic with username=installer.

Yesterday my PW firmware was updated from 1.49.0 to 1.50.1 and my code fails at login with the response:
\{"code":401,"error":"roles [Provider_Engineer] require physical access","message":"Login Error"\}

There is also a new screen when installer is selected in the Kiosk.  It states that the PW switch must be toggled (as proof of physical access, I presume).

Any ideas?

spoonwzd, 2020-10-23:
I too had this issue yesterday when the new firmware went on. I don't think there's a way around it using the local gateway now they have introduced physical confirmation.

I've switched my code over to using the public API now, which actually seems to work better as the process doesn't stop the Powerwall from operating when making a change like with the local gateway. Changes to reserve seem to be instant.

https://www.teslaapi.io/powerwalls/commands

All you need to do is get the token from /oauth/token then you can set the reserve % via /api/1/energy_sites/:siteid/backup

BJReplay, 2020-10-23:
@spoonwzd can you share how you do this?  I assume that the reserve percentage is a parameter on the call?  The API doco doesn't have this detail.  e.g. is it /api/etc/backup?reserve=10 or api/etc/backup/reserve/10?

HeraldCoupe, 2020-10-23:
This is so annoying but perhaps inevitable.

spoonwzd, 2020-10-23:
You simply POST the backup_reserve_percent value in JSON to the /api/1/energy_sites/energy_site_id/backup API URL along with the authorization bearer @BJReplay 

This is how I'm doing it in node-red with a custom JavaScript function:

msg.headers = {
    Authorization: "Bearer " + MY_AUTH_TOKEN,
    "content-type": "application/json"
}
msg.method = "POST"
msg.url = "https://owner-api.teslamotors.com/api/1/energy_sites/MY_SITE_ID/backup"
msg.payload = {
    "backup_reserve_percent": 0
}

HeraldCoupe, 2020-10-23:
> You simply POST the backup_reserve_percent value in JSON to the /api/1/energy_sites/energy_site_id/backup API URL along with the authorization bearer @BJReplay
> 
> This is how I'm doing it in node-red with a custom JavaScript function:
> 
> msg.headers = {
> Authorization: "Bearer " + MY_AUTH_TOKEN,
> "content-type": "application/json"
> }
> msg.method = "POST"
> msg.url = "https://owner-api.teslamotors.com/api/1/energy_sites/MY_SITE_ID/backup"
> msg.payload = {
> "backup_reserve_percent": 77
> 

I don't suppose you've got something similar for changing modes have you?

spoonwzd, 2020-10-23:
> I don't suppose you've got something similar for changing modes have you?

I haven't tried changing the mode, but I'm going to assume it's just the same as the above example, just use /api/1/energy_sites/energy_site_id/operation and set "mode" in the JSON payload with "self_consumption" or whatever you're looking to set.

BJReplay, 2020-10-23:
Ahh, so annoying.  I'm testing using postman and I'm getting 

```
{
    "response": null,
    "error": "https://powergate.prd.sn.tesla.services:443/api/powerwalls/MY_GATEWAY_ID/backup => Method Not Allowed",
    "error_description": ""
}
```

In response to:

```
Method: POST
URL: https://owner-api.teslamotors.com/api/1/powerwalls/MY_SITE_ID/backup 
Headers: as noted above (Bearer & auth token, content-type)
Body: {"backup_reserve_percent": 10}

```

spoonwzd, 2020-10-23:
Is it that you need to be setting reserve at 'energy_sites' not 'powerwalls' @BJReplay ?

BJReplay, 2020-10-23:
@spoonwzd Yes!  Thanks!  Right.  powerwalls for getting status (or maybe that works with energy_sites too - will try), energy_sites for setting status.

spoonwzd, 2020-10-23:
> @spoonwzd Yes! Thanks! Right. powerwalls for getting status (or maybe that works with energy_sites too - will try), energy_sites for setting status.

Nice. :) I'm still getting live power flow stats from the local gateway as I don't want to be hammering the public API.

BJReplay, 2020-10-23:
@HeraldCoupe Yes, I have:

POST to https://owner-api.teslamotors.com/api/1/energy_sites/YOUR_INTEGER_SITE_ID/operation
Headers:
Authorization Bearer as above
Content-Type application/json
Body:

`{"default_real_mode":"self_consumption"}`
or
`{"default_real_mode":"backup"}`

spoonwzd, 2020-10-23:
Great work everyone :)

HeraldCoupe, 2020-10-23:
Yes, great work and thank you. I'll have a go at this myself in the coming days.
The next question is.... how long before Tesla stop us from using this API too?

HeraldCoupe, 2020-10-24:
> @HeraldCoupe Yes, I have:
> 
> POST to https://owner-api.teslamotors.com/api/1/energy_sites/YOUR_INTEGER_SITE_ID/operation
> Headers:
> Authorization Bearer as above
> Content-Type application/json
> Body:
> 
> `{"default_real_mode":"self_consumption"}`
> or
> `{"default_real_mode":"backup"}`

Brilliant, thank you.
I've made progress today, I can get the token and energy_site_id so I'll move onto getting the mode then I'll attempt to set the mode.

HeraldCoupe, 2020-10-25:
You can use "https://owner-api.teslamotors.com/api/1/energy_sites/YOUR_INTEGER_SITE/site_info" to get "backup_reserve_percent", "default_real_mode" as well as "storm_mode_enabled" and a bunch of other potentially useful bits of info. 
It requires the usual authorization bearer stuff.

BJReplay, 2020-10-25:
Note that the site ID is not an integer (as I suggested by referring to it as YOUR_INTEGER_SITE) - more like a BIGINT or long numeric.  I've seen an example which is 12 digits long.

fkhera, 2020-11-03:
Any idea how to force the "commit" for the setings.
In general from what I see calling the cloud API , is introducing around 15-20 minute delay, closer to 15 minute.

Any way for the settings to take effect.
The Application shows the setting right away, so its going to Tesla server, but the execution of Gateway is taking around 15 minutes.

Is there a way to call ?
/config/completed  or /sitemaster/run from the CLOUD?

https://github.com/fkhera/powerwallCloud/blob/master/powerwallBackup.py#L32

HeraldCoupe, 2020-11-03:
Interesting. I'd expected to have this problem too but actually the change appears to be pretty instant for me. I've confirmed it using the graph in the Tesla app.

spoonwzd, 2020-11-03:
> Is there a way to call ?
> /config/completed or /sitemaster/run from the CLOUD?

I don't change mode - simply adjusting the reserve fulfills all my needs.

When using the public API confirmation (/config/completed or /sitemaster/run) is not required and the change is all but instant on my Powerwall.

fkhera, 2020-11-03:
I just stay in the time based control mode, and adjust reserve %.
Today I noticed there is about 10-15 minute delay on my phone.

Our current peak hours are 5am-9am and 5pm-9pm.
One long peak defined as 5am-9pm in Tesla peak and controlled by adjusting
reserves.
This morning I am supposed to be at zero %, but I had another script set me
to 100% at around six to 6:50am.
At 6:50 am on my phone , I manually adjusted from 100% to force discharge 0%
We did not see change of standby to 'discharging' until around 7:05am.
Also seeing around 15 minute delay.

So I guess the delay matches the phone so must be Tesla server or Tesla
Gateway related.

spoonwzd, 2020-11-03:
I wonder if it's something to do with the mode you're in? I'm in self_consumption mode and never switch away from it - I just set the battery reserve and the change is actioned on the Powerwall instantly.

burble61, 2020-11-26:
> I wonder if it's something to do with the mode you're in? I'm in self_consumption mode and never switch away from it - I just set the battery reserve and the change is actioned on the Powerwall instantly.

Thanks for the insight. I tried this - for Time Based Control, looking at a separate energy monitor (ie. not relying on the Tesla app) changes took between 15 and 45 minutes to take effect. Self consumption - as you experienced - pretty  immediate.

Discovered another factor - under TBC and Powerwall controlled charging, charge power is consistently c.3.8KW.  When charging due to increased reserve % under TBC - 5KW. Understandable to an extent, to ensure the reserve is put in place asap.  But charging under Self Consumption forced by increased reserve.....1.7KW, consistent over a number of tests.

Any views why that might be (and how, potentially, to increase it)? I ask the question as although SC seems to give the control response needed, that charge rate is just too low for my scenario.

BJReplay, 2020-11-26:
> Any views why that might be (and how, potentially, to increase it)?

Yep, setting the mode to backup will double the charge rate to 3.3kW.  You will need to monitor progress, and when you hit your target you'll need to swap back into SC.

spoonwzd, 2020-11-26:
I can confirm that the max charge rate in self consumption mode is 1.7kW and there's no way to increase it when in this mode.

[image]

Luckily I have a 10-hour off-peak window so I can achieve a full charge within this time frame.

burble61, 2020-11-30:
> > Any views why that might be (and how, potentially, to increase it)?
> 
> Yep, setting the mode to backup will double the charge rate to 3.3kW. You will need to monitor progress, and when you hit your target you'll need to swap back into SC.

In backup charge rate on my unit is 5KW. But agree with the approach you suggest - i'll test the required charge against available time and if the SC rate can't meet that, its a simple calculation to determine the charge profile in the off peak period.

Can understand why backup charges at the full 5KW rate, I have asked my installer why the rates are different in SC/TBC. Will share any response.

burble61, 2020-12-09:
I did query Tesla on the difference in charging rates (1.7KW in Self Consumption, 5KW in Backup, 3.7KW in TBC).

(Eventual) response was that 3.7KW was chosen as a balance between charge rate & battery lifetime....but that 1.7KW in SC couldn't (wouldn't ?) be changed. Go figure.

RichardHill, 2020-12-10:
Firstly I would like to thank everyone on this thread for how instructional this has been - its been brilliant.

Secondly - and this may not be apparent to others so I thought I would share. If you have a tariff that does not differ between night and day (in the UK we call this Economy7) - then you cannot set the reserve. This is what I see on the Octopus Tesla Tariff -: 

[image]

Thats after calling -: 

https://owner-api.teslamotors.com/api/1/energy_sites/{site-id-number}/backup

with a body of -:

{ "backup_reserve_percent" : 50 }

from Postman. (ps I've removed my site-id and replaced it with {site-id-number}.

fkhera, 2020-12-10:
Can you set reserve in your phone ??

RichardHill, 2020-12-10:
@fkhera  - no - its been removed. 

I noticed that when I check the Data end point a few settings are set to false eg -:

Under components -: 

```
"backup" : false
"configurable": false,
```

and

```
        "backup": {
            "backup_reserve_percent": 0,
            "events": null
        },
```

vloschiavo, 2021-02-21:
closed.  See issue #41 for an update re: version 20.49.0

closed 2021-02-21 by vloschiavo