vloschiavo/powerwall2 #35: Force Update/Commit on the Cloud API

fkhera, 2020-11-03:
Any idea how to force the "commit" for the setings.
In general from what I see calling the cloud API , is introducing around 15-20 minute delay, closer to 15 minute.

Any way for the settings to take effect.
The Application shows the setting right away, so its going to Tesla server, but the execution of Gateway is taking around 15 minutes.

Is there a way to call ?
/config/completed or /sitemaster/run from the CLOUD?

https://github.com/fkhera/powerwallCloud/blob/master/powerwallBackup.py#L32

BJReplay, 2020-11-03:
Most (all) people have reported near enough to an instantaneous change in operations since late 1.4x versions.  I know that I'm getting instantaneous changes on 1.50.1 via cloud API.

Are you setting the mode to Advanced - Time-based control - it looks like it from https://github.com/fkhera/powerwallCloud/blob/master/powerwallBackup.py#L48 "autonomous"?  That might be the difference.  Changes to Self-powered ("self_consumption") or backup seem to work instantly.

fkhera, 2020-11-03:
Do you mean change the mode as well as the percent ?
We just change percent would change mode make a difference ?

BJReplay, 2020-11-03:
It's hard to say - you haven't really described what you mean by "is introducing around 15-20 minute delay, closer to 15 minute."

So, from "We just change percent", I'm guessing that you change the percentage reserved for backup, and from "https://github.com/fkhera/powerwallCloud/blob/master/powerwallBackup.py#L32" I'm guessing you do it in autonomous mode.

I'm now guessing that you run your powerwall in autonomous mode, and you change the percentage reserved for backup, but you don't observe it start to charge or discharge as you are expecting immediately, but it takes around 15 minutes.

My guess is that the powerwall is spending some time trying to work out what it should do - does it expect to be able to reach your new backup target based on current activity, or does it need to change.

What happens if you make exactly the same change in the phone app?  Does it change behavior instantly, or after about 15 minutes?

fkhera, 2020-11-03:
Yes you are right . Usually we see similar delay on phone .

fkhera, 2020-11-03:
In cloud api setting mode independent from reserve
 Percent . In local api this is same rest call. I just set percent reserve
does this cause delay?

BJReplay, 2020-11-03:
I think you've answered your own question - you see a similar delay making changes to backup reserved percentage when you leave the mode in autonomous mode on your phone as when you use the API.

closed 2021-02-21 by vloschiavo