vloschiavo/powerwall2 #39: Power history

jcwright77, 2021-01-22:
Any idea if there is an endpoint for power history ? Or is this stored elsewhere ?

vloschiavo, 2021-02-02:
I haven't seen one...You're talking about what is in the Tesla app?  I'm
pretty sure they are storing your stats on the tesla servers and presenting
that to the app.  I do something similar with OpenHab and Home Assistant.
 Collect the stats and graph.

vloschiavo, 2021-02-02:
[image: image.png]

[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

jcwright77, 2021-02-02:
ok, thanks. I figured it was in the cloud, was hoping for a public api but it seems not. I will try to archive going forward as you suggest.
-john wright

closed 2021-02-02 by jcwright77