vloschiavo/powerwall2 #43: PW internal temperature?

jh125486, 2021-02-16:
Anybody come across an endpoint to the PW internal temperature?  

I'm having to put heat lamps on my PWs to keep them within range, and tech support wouldn't give me the API they are using to pull that internal info.

emeins, 2021-02-16:
I am not aware of any API that delivers the PW temperature. I found something but only for cars: https://www.openhab.org/addons/bindings/tesla/
I added an additional Zigbee temp sensor and monitor it.

jh125486, 2021-02-16:
> I am not aware of any API that delivers the PW temperature. I found something but only for cars: https://www.openhab.org/addons/bindings/tesla/
> I added an additional Zigbee temp sensor and monitor it.

Thanks for that idea.  
I've screwed an extra Ecobee sensor into my garage wall near the PWs, so I'll at least have the ambient temp.

closed 2021-02-16 by jh125486