vloschiavo/powerwall2 #45: POST to api/operation fails: User does not have adequate access rights

terafin, 2021-02-26:
On the most recent firmware - this command: 
`curl -k --header "Authorization: Bearer SK_MY_BEARER_TOKEN==" -X POST -d '{"mode":"self_consumption","backup_reserve_percent":24.6}' https://THE_IP/api/operation`

Seems to fail with this error:
```
{"code":403,"error":"Unable to POST to resource","message":"User does not have adequate access rights"}%
```

It worked a while ago (as I was using this in `http://github.com/terafin/mqtt-tesla-bridge`), but it seems quite dead now.

I've tried various forms of installer logins (resetting password on GW, etc). Am I missing something, has anyone had any success with POSTs to change mode?

BJReplay, 2021-02-27:
See discussion at #34 

Control needs to be via the cloud API - which is itself problematic at the moment.

terafin, 2021-02-27:
@BJReplay thank you! Yikes... that's less than stellar. I'll follow along there. Cheers!

BJReplay, 2021-02-27:
@terafin make sure you follow the discussions at https://github.com/timdorr/tesla-api/discussions/ and the instructions at https://tesla-api.timdorr.com/ but note that Tesla seem to be making it harder and harder to use the cloud API.

My controller is using Tim Hollander's auth library https://github.com/tomhollander/TeslaAuth and is currently able to renew a token, but new login attempts fail.

terafin, 2021-02-27:
Note this seems to work, when I nabbed the installer cookie from logging in as an installer (and toggling the gateway)...
```
curl -k --cookie "AuthCookie=XXXXXX==; UserRecord=XXXXXXXXYYYY=="  -X POST -d '{"mode":"self_consumption","real_mode":"self_consumption","backup_reserve_percent":24.6}' https://MY_IP/api/operation
```

terafin, 2021-02-27:
Will watch for how long this auth cookie lasts for...

terafin, 2021-02-27:
Looks like around 10 hours or so

McBoerny, 2021-02-28:
The following works for me (example using session cookie):
```
curl -k -O -c cookies.txt -X POST https://IP-Powerwall/api/login/Basic -H "Content-Type: application/json" 
-d "{\"username\": \"customer\",\"email\": \"your.login@mail.address\",\"password\": \"yourPassword\"}"

curl -k -O -b cookies.txt https://IP-Powerwall/api/meters/aggregates
```
Note: I had to reset the Powerwall and enter the new customer password to get it working, but YMMV.

terafin, 2021-03-01:
Yup, works great for that request, but doesn’t seem to work for a POST to the operation endpoint

BJReplay, 2021-03-01:
> Yup, works great for that request, but doesn’t seem to work for a POST to the operation endpoint

yeah, customer login allows you to stop and start, but not change modes.  Still need installer mode (which still needs a physical switch toggle) for that - so you're really forced to use the telsa (cloud) API that the phone apps use.

closed 2021-08-06 by vloschiavo