vloschiavo/powerwall2 #46: This isn't a real issue, more like a collaboration request! :dagger:

Viss, 2021-03-17:
Hi!
 I have two powerwalls hooked to one tesla backup gateway 2, and I do have plans to setup a transfer switch so that I can have a generator. I'm in the US and my system runs 240 volt back to front, but sadly my generator is only 120v, so I need a new one - but I am keenly interested in two things, specifically

1) Any intel you/we/I can get about messing with the generator component

2) if its possible to manually twiddle bits to enable 'stormwatch mode'.

Stormwatch mode is where the system believes there's horrible weather coming, and will flip to charging the batteries from the grid until the weather services say its all clear (in my part of San Diego, this happened four times in december, so I got to see first hand).

In theory, this would work:
1) get 240 volt generator (the honda  7000 watt one is the one I'm targeting)
2) have a transfer switch installed between the backup gateway and the grid - the intention here would be to replace the grid with the generator
3) somehow flip the backup gateway into stormwatch mode, thereby shunting any available power beyond what the house is consuming from the generator into the batteries. 
4) probably bake some cookies in the electric toaster oven :D

Let me know if you want to screw around!

vloschiavo, 2021-03-17:
Hey,

Yes, that makes sense.  I'm not sure if stormwatch mode is exposed via the
local API (at least I haven't found it yet).  However, you should be able
to change the reserve percentage to 100%.  This would, effectively, put the
powerwalls into 'charge' only mode and you can bake cookies while on
generator power. ;)

Then, once your cookies are finished baking and you want to turn off your
generator, you can change the reserve percentage to something lower, e.g.
15% and your powerwalls will continue to run the house off battery.

There's another issue (#19) filed where folks were playing with
automatically changing the reserve percentage to change when (and for how
long) to force the powerwalls to charge vs discharge and vice versa.   Also
check out issue #20 as well.

-V

karelwh, 2021-03-17:
> Cool!
> I'll start to play.
> I have to make some network changes (creating an iot wlan, vlans, natting
> etc) before i can dump the powerwall on my network - I have two actual
> powerwalls, which is nice, because their combined output lets me run the
> entire house off the things (even the hot tub!) - the concept with
> twiddling stormwatch is to basically force the system to charge the
> batteries from the generator. The breville toaster ovens are only like 900
> watts, so they barely make a dent in the battery - but the big appliances,
> electric dryer, electric oven, are 5kw devices, and those things put
> huuuuge dents in the battery. I've learned a lot about this stuff since
> october, heh

Hi

The following two links may be of interest to you:
https://www.tesla.com/en_AU/off-grid?redirect=no
https://www.solarquotes.com.au/blog/tesla-powerwall-off-grid/

They show that back up with a generator (they have approved two models) is
allowed in a totally off grid system, in Australia at least.

regards
Karel

ps thanks for your work Vince, the authentication to the API had me beat.

closed 2021-08-06 by vloschiavo