vloschiavo/powerwall2 #48: new authentication requirements

dustin, 2021-07-04:
My software updated and apparently basic status requires authentication now.

jorythompson, 2021-07-26:
I just got powerwall2 installed and am having authentication issues.  I got the pem file like this:
```echo quit | openssl s_client -showcerts -servername powerwall -connect powerwall:443 > cacert.pem```

I tried this:
```curl --cacert cacert.pem -k -s -i -X POST -H "Content-Type: application/json" -d '{"username":"customer","password":"last-5-of-gateway-serial","email":"my email","force_sm_off":false}'  https://powerwall/api/login/Basic```

and got back the bad credentials error:
```
HTTP/2 401
content-type: application/json
x-content-type-options: nosniff
content-length: 62
date: Mon, 26 Jul 2021 14:47:47 GMT                                                                                                                                                                                                             {"code":401,"error":"bad credentials","message":"Login Error"}```

I've tried my password instead of the last 5 characters of the gateway, but it fails as well.
I am able to log into the gateway via a browser

darryllee, 2021-08-05:
Hey there @jorythompson - I had success with this:
`curl -c cookies.txt -b cookies.txt -s -k -X POST "https://$GATEWAY/api/login/Basic" -H "Content-Type: application/json" -d '{"username":"customer","password":"$PASSWORD","email":"$EMAIL"}'`

(I've messed with the cert and have not had any luck. So I'm just using -k to ignore the cert mismatch.)

This logs me in and sets a cookie that I can then use for subsequent commands (by specifying the cookies.txt file).

I've wrapped these up into some rudimentary logging scripts here: 
https://github.com/darryllee/teslog

Wanted to note that I couldn't have done any of this without @vloschiavo 's work. Thanks!

closed 2021-08-06 by vloschiavo

darryllee, 2021-08-06:
As for the cert issue, I've tried @vloschiavo's technique to grab the cert, as well as exporting it directly from Chrome (which gave me the same thing). But I tried referencing it using --cacert as well as --capath (putting it in a directory), but every time, I got:
```
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html
```
I wonder if the problem is that the cert isn't for a CA, but just for the host. Or is the CA also the server cert? Is that allowed? Here's the first bit of my openssl output:
```
Certificate chain
 0 s:C = US, ST = California, L = Palo Alto, O = Tesla, OU = Tesla Energy Products, CN = eba138a4a64f427ea6344a9002a1723d
   i:C = US, ST = California, L = Palo Alto, O = Tesla, OU = Tesla Energy Products, CN = eba138a4a64f427ea6344a9002a1723d CA
```