vloschiavo/powerwall2 #49: /meters/aggregates 404 on FW 21.20.3

kairojya, 2021-07-04:
I'm getting a 404 response from /meters/aggregates since the 21.20.3 update. Many of the other end points I checked are responding. I've seen others post on Reddit and TMC about the /vitals page no longer working from the web interface of their gateway, perhaps this is related?

fkhera, 2021-07-04:
Slowly tesla just taking all the endpoints away. Does authentication work ?

kairojya, 2021-07-04:
> Slowly tesla just taking all the endpoints away. Does authentication work ?

Are you talking about the POST to [/api/login/basic](https://github.com/vloschiavo/powerwall2#post-apiloginbasic)? I haven’t tried it. I’ve only had solar and Powerwalls for a week and have been experimenting a bit before PTO.

What I’ve done is to log in by just hitting my gateway/inverter by its IP on my home’s network (https://192.168.X.X). That prompts me with the Installer/Customer login form. I was able to login to it as a customer on 21.20.2 and 21.20.3. After logging in, I’ve been able to reach various endpoints just using a browser. I’ve tried curl (win10 WSL2) with the -k switch but got the ‘authentication required’ response.

I have Tesla installed solar panels (8.16 kW), a Powerwall Plus (3012170-25-B), Powerwall 2 (3012170-05-B), and Backup Gateway 2. Let me know if you’d like me to try something.

fkhera, 2021-07-05:
Sounds like it’s just token authentication so you would have to resolve to
get token first then make future calls and refresh token as needed

kairojya, 2021-07-08:
I tried the end point again today and it worked. I’m not sure why it wasn’t working the other day when I opened this. I tried it (/meters/aggregates) right after trying other end points, and it was the only one responding with 404.

closed 2021-07-08 by kairojya