vloschiavo/powerwall2 #50: Access restricted to TEG network?

holman, 2021-07-13:
I just got a new Powerwall installed a bit ago and it looks like I'm on a newer firmware version, too (21.20.3).

I haven't played with a Powerwall before, so forgive me if I'm wrong, but it looks like historically you could access the Gateway directly by hitting its local IP address on your own network, yeah? I'm not able to do that- I can only access the Gateway from its own TEG-XXX wifi network.

Mostly was worried that this is a new change and they're going the route of the Wall Connector, which has also recently-ish been locked down to local-network-access-only. Would be a bummer.

BJReplay, 2021-07-13:
I'm on 21.20.2, and can hit it on the IP address assigned by my router - but only on https (it presents a dodgy self-signed cert).

No redirect, no response on http.

I'm guessing you just tried http?

If not, confirmation that what worked on 21.20.2 no longer works on 21.20.3

holman, 2021-07-14:
Yeah- been trying it over http and https. No response on either protocols.

BJReplay, 2021-07-14:
Uggh, that's annoying.  I'm hitting it locally for monitoring (https://powerwall/api/meters/aggregates) to collect data every six seconds.  I'll have to point that at the public API that the app uses, and then risk being hit with a captcha.

tj1816, 2021-07-14:
Try reconnecting the powerwall to your home network. Even if it says it's connected, just click on the network name and it will connect again.  I have a problem where about every week I have to reconnect it otherwise I can't access it from my home network. Has anyone else run into that problem?

holman, 2021-07-14:
PEBKAC!

It was on a different wifi network than I had originally thought it was. Your suggestion to reconnect made me recheck that, so thanks. Everything's working now, and it does indeed look like Powerwalls are still accessible on the local network (...for now!)

closed 2021-07-14 by holman