vloschiavo/powerwall2 #51: Found a new endpoint, but it returns a binary file

darryllee, 2021-08-06:
https://teg/api/devices/vitals

Good old **strings** returns some interesting tidbits:
`        THC_State*
THC_STATE_AUTONOMOUSCONTROL
THC_AmbientTemp!
`
I'd love to get temperature info from my gateway/inverter/battery since I was having problems with my brand-new Tesla 7.6kW inverter because it was installed with low coolant, and until it got topped off last week, was throwing "Inverter Over Temperature" alerts (which after a recent update, were only visible after logging on as Installers instead of Customers. Super-shady.)

vloschiavo, 2021-08-06:
Nice find!  (And nice work on your project https://github.com/darryllee/teslog).  

Feel free to issue a pull request with what you know so far, I'm sure the rest of the community will be interested and can help learn more.

Thanks!

darryllee, 2021-08-06:
DOH, I just saw you already published a bash script that's far more elegant than mine. I will create a PR for the vitals endpoint I discovered. :-}

vloschiavo, 2021-08-06:
I'm taking a look at the binary file output on my powerwall now.  It looks like the various serial numbers of the gateway and the battery units are plastered all over this file.  So I suggest you scrub those out before posting an example.

I don't know if there is some special encoding or not yet. But I suspect that these are key-value pairs and probably some standard format from Redis or something similar.    I'll put into a hex editor and see what I can reverse engineer.

example:  I'm looking at the output in VIM:

Line 24: 
^OTHC_AmbientTemp!^@^@^@^@^@^@=@

THOC_AmbientTemp is likely the key, and the value is encoded in the hexidecimal value here: ^@^@^@^@^@^@=@ ;  I'll need to see the hex values vs the ASCII translation to see if the numbers make sense.

These would be interesting to have the values too:
 37 ^XPOD_nom_energy_remaining!^@^@^@^@^@~Á@^R#
 38 ^XPOD_nom_full_pack_energy!^@^@^@^@^@×Ê@^R%
 39 ^ZPOD_available_charge_power!^@^@^@^@^@X»@^R%
 40 ^ZPOD_available_dischg_power!^@^@^@^@^@X»@^R%
 41 ^ZPOD_curtailed_charge_power!^@^@^@^@^@^@^@^@^R%
 42 ^ZPOD_curtailed_dischg_power!^@^@^@^@^@^@^@^@^R^_

Also, setting up a "charge completed" notification would be useful in my automation to then enable charging for the car on those days when I'm making more solar than I can use/store in the powerwall.  But it looks like these are broken out for each powerwall (if you have multiples on the same gatway).

Powerwall 1:
Line 50:
 50 ^RPOD_ChargeComplete0^@^R^Y

Others:
 51 ^UPOD_DischargeComplete0^@^R^[
 52 ^WPOD_PersistentlyFaulted0^@^R^Z
 53 ^VPOD_PermanentlyFaulted0^@^R^U
 54 ^QPOD_ChargeRequest0^@^R^U
 55 ^QPOD_ActiveHeating0^@^R^O
 56 ^KPOD_Resting0^@^R^O
 57 ^KPOD_CCVhold0^@^R^Z
 58 ^VPOD_HeatZeroCurrentReg0^@^R^\
 59 ^XPOD_ExtendedCapacityMode0^@^R^Y
 60 ^UPOD_IpcRequestedBusUp0^@^R^Y
 61 ^UPOD_SelfTestRequested0^A

Powerwall 2:
134 ^RPOD_ChargeComplete0^@^R^Y
135 ^UPOD_DischargeComplete0^@^R^[
136 ^WPOD_PersistentlyFaulted0^@^R^Z
137 ^VPOD_PermanentlyFaulted0^@^R^U
138 ^QPOD_ChargeRequest0^@^R^U
139 ^QPOD_ActiveHeating0^@^R^O
140 ^KPOD_Resting0^@^R^O
141 ^KPOD_CCVhold0^@^R^Z
142 ^VPOD_HeatZeroCurrentReg0^@^R^\
143 ^XPOD_ExtendedCapacityMode0^@^R^Y
144 ^UPOD_IpcRequestedBusUp0^@^R^Y
145 ^UPOD_SelfTestRequested0^A

This is also interesting:
172 ^SPINV_MainRelayState*^LRELAY_CLOSED^R&
173 ^VPINV_NeutralRelayState*^LRELAY_CLOSED^R^V

darryllee, 2021-08-06:
Ha yes, I just created a PR **without** including my vitals file because of it contained my serial numbers, and I didn't feel like loading up a hex editor and scrubbing that info.

But yes, tons of very interesting info that I've not seen in other endpoints.

vloschiavo, 2021-08-06:
Not sure if I'm going the right direction but I'm looking at this:
```THC_AmbientTemp!......<@```

0x0A (Hex value 0A) plus another hex byte always precedes each Value.  (maybe 0x0A indicates the end of the line?)

The ! = 0x21 - This looks like a the delimiter for the end of each of the keys.
The @ = 0x40 - This also looks like a delimiter as well.

So the hex in between those delimiters should be the value.
In my case I see the following HEX as the value:
```00 00 00 00  00 80 3C```

0x80  = 128
0x3C = 60

Does this mean 128.60 °F ?

Seems a bit hot.  My ambient outside of the chassis is 80.24 °F.

vloschiavo, 2021-08-06:
> Ha yes, I just created a PR **without** including my vitals file because of it contained my serial numbers, and I didn't feel like loading up a hex editor and scrubbing that info.
> 
> But yes, tons of very interesting info that I've not seen in other endpoints.

Merged.

darryllee, 2021-08-07:
Here's a bit of data between **THC_AmbientTemp!** up to **" TEPOD**:

At 2:36PM (ambient temp 88ºF)
```00 00 00 00 40 42 40 0A EF 06 0A 97 01 0A 94 01 0A```

At 6:06PM (82ºF)
```CE CC CC CC CC 4C 42 40 0A 80 07 0A 97 01 0A 94 01 0A```

At 6:26PM (80ºF)
```9A 99 99 99 99 19 42 40 0A 80 07 0A 97 01 0A 94 01 0A```

So this seems consistent for end of data:
```0A 97 01 0A 94 01 0A```

darryllee, 2021-08-07:
From user [power.saver](https://teslamotorsclub.com/tmc/members/power-saver.72915/) on the [teslamotorclub.com forums](https://teslamotorsclub.com/tmc/threads/discussion-how-to-get-status-information-from-your-powerwalls.191898/page-4#post-5788939): 

> The binary data after the text strings is the value, in 64-bit IEEE-754 Floating Point notation. There is a ! character separating the text from the number, and the 8 bytes are in little-endian format.
>
> For example, in the vitals file is the text:
> `THC_AmbientTemp!`
> 
> This is followed by 8 bytes (shown here as hexadecimal digits):
> `CE CC CC CC CC CC 40 40`
>
> Here they are in big-endian format (again shown as hexadecimal digits):
`40 40 CC CC CC CC CC CE`
>
> Those 8 bytes comprise the 64-bits of the IEEE-754 Floating Point number.
>
> The decimal equivalent of that is 33.6 which is the temperature in Celsius. (92.5°F)
>
>
> Text fields that are followed by a * character have a text value after them (enumeration).
> 
> The object:
> `THC_State*`
> has the value:
> `THC_STATE_AUTONOMOUSCONTROL`
> 
> An interesting one in this file is:
> `POD_DC_Bus_Voltage!`
> which has a value of 428.1 volts.

sean-user, 2021-08-08:
I'm [power.saver](https://teslamotorsclub.com/tmc/members/power-saver.72915/) on the TMC forums.

Some additional info on the objects with * after them (text enumerations). The first byte after the * character is the length of the text string, followed by the actual text. That wasn't explained in my post on the TMC forums.

sean-user, 2021-08-08:
With a little work, I was able to parse out most of the vitals file. Here is the text result. The first column is the number of bytes into the file where the parser found the start of an object. This will probably vary since differences in the text string lengths affects the file size.

**A new delimiter was found**: if a 0 (ascii zero) appears after the object then the value is TRUE or FALSE, and is represented by a binary 1 or 0 in the next byte. I show TRUE or FALSE when this occurs.

```
0031 : 1118431-00-J = 
00b4 : 1092170-03-G = 
0128 : THC_State = THC_STATE_AUTONOMOUSCONTROL
0152 : THC_AmbientTemp = 33.600000
019b : 1081100-79-J = 
020b : POD_nom_energy_to_be_charged = 2756.000000
0234 : POD_nom_energy_remaining = 11115.000000
0259 : POD_nom_full_pack_energy = 13707.000000
027e : POD_available_charge_power = 7000.000000
02a5 : POD_available_dischg_power = 7000.000000
02cc : POD_curtailed_charge_power = 0.000000
02f3 : POD_curtailed_dischg_power = 0.000000
031a : POD_est_DC_Bus_Power = -2320.000000
033b : POD_Ibatt = -50.600000
0351 : POD_Vbatt = 46.600000
0367 : POD_DC_Bus_Voltage = 428.100000
0386 : POD_state = POD_ACTIVE
039f : POD_enable_line = TRUE
03b4 : POD_ChargeComplete = FALSE
03cc : POD_DischargeComplete = FALSE
03e7 : POD_PersistentlyFaulted = FALSE
0404 : POD_PermanentlyFaulted = FALSE
0420 : POD_ChargeRequest = FALSE
0437 : POD_ActiveHeating = FALSE
044e : POD_Resting = FALSE
045f : POD_CCVhold = FALSE
0470 : POD_HeatZeroCurrentReg = FALSE
048c : POD_ExtendedCapacityMode = FALSE
04aa : POD_IpcRequestedBusUp = FALSE
04c5 : POD_SelfTestRequested = TRUE
050e : 1081100-79-J = 
057e : PINV_EnergyDischarged = 4173130.000000
05a0 : PINV_EnergyCharged = 4827370.000000
05bf : PINV_VSplit1 = 123.300000
05d8 : PINV_VSplit2 = 123.200000
05f1 : PINV_PllFrequency = 59.995000
060f : PINV_PllLocked = TRUE
0623 : PINV_Pout = 2.320000
0639 : PINV_Qout = -0.020000
064f : PINV_Vout = 246.500000
0665 : PINV_Fout = 60.011000
067b : PINV_ReadyForGridForming = TRUE
0699 : PINV_State = PINV_GridFollowing
06bb : PINV_GridState = Grid_Compliant
06dd : PINV_MainRelayState = RELAY_CLOSED
0702 : PINV_NeutralRelayState = RELAY_CLOSED
072a : PINV_Iavail = 32.400000
0742 : PINV_Savail = 7.990000
075a : PINV_Vdc = 428.100000
076f : PINV_HardwareEnableLine = TRUE
078c : PINV_PowerLimiter = PWRLIM_No_Power_Limit
080d : 1092170-03-G = 
0881 : THC_State = THC_STATE_AUTONOMOUSCONTROL
08ab : THC_AmbientTemp = 29.100000
08f4 : 1081100-59-J = 
0964 : POD_nom_energy_to_be_charged = 2758.000000
098d : POD_nom_energy_remaining = 11207.000000
09b2 : POD_nom_full_pack_energy = 13810.000000
09d7 : POD_available_charge_power = 7000.000000
09fe : POD_available_dischg_power = 7000.000000
0a25 : POD_curtailed_charge_power = 0.000000
0a4c : POD_curtailed_dischg_power = 0.000000
0a73 : POD_est_DC_Bus_Power = -2350.000000
0a94 : POD_Ibatt = -51.200000
0aaa : POD_Vbatt = 46.600000
0ac0 : POD_DC_Bus_Voltage = 428.000000
0adf : POD_state = POD_ACTIVE
0af8 : POD_enable_line = TRUE
0b0d : POD_ChargeComplete = FALSE
0b25 : POD_DischargeComplete = FALSE
0b40 : POD_PersistentlyFaulted = FALSE
0b5d : POD_PermanentlyFaulted = FALSE
0b79 : POD_ChargeRequest = FALSE
0b90 : POD_ActiveHeating = FALSE
0ba7 : POD_Resting = FALSE
0bb8 : POD_CCVhold = FALSE
0bc9 : POD_HeatZeroCurrentReg = FALSE
0be5 : POD_ExtendedCapacityMode = FALSE
0c03 : POD_IpcRequestedBusUp = FALSE
0c1e : POD_SelfTestRequested = TRUE
0c67 : 1081100-59-J = 
0cd7 : PINV_EnergyDischarged = 4179150.000000
0cf9 : PINV_EnergyCharged = 4857010.000000
0d18 : PINV_VSplit1 = 123.400000
0d31 : PINV_VSplit2 = 123.100000
0d4a : PINV_PllFrequency = 60.007000
0d68 : PINV_PllLocked = TRUE
0d7c : PINV_Pout = 2.330000
0d92 : PINV_Qout = -0.030000
0da8 : PINV_Vout = 246.600000
0dbe : PINV_Fout = 60.013000
0dd4 : PINV_ReadyForGridForming = TRUE
0df2 : PINV_State = PINV_GridFollowing
0e14 : PINV_GridState = Grid_Compliant
0e36 : PINV_MainRelayState = RELAY_CLOSED
0e5b : PINV_NeutralRelayState = RELAY_CLOSED
0e83 : PINV_Iavail = 32.400000
0e9b : PINV_Savail = 7.980000
0eb3 : PINV_Vdc = 428.100000
0ec8 : PINV_HardwareEnableLine = TRUE
0ee5 : PINV_PowerLimiter = PWRLIM_No_Power_Limit
0f67 : 1118431-00-J = 
0fdb : ISLAND_VL1N_Main = 122.500000
0ff8 : ISLAND_FreqL1_Main = 60.010000
1017 : ISLAND_VL1N_Load = 122.500000
1034 : ISLAND_FreqL1_Load = 60.010000
1053 : ISLAND_PhaseL1_Main_Load = 0.000000
1078 : ISLAND_VL2N_Main = 121.000000
1095 : ISLAND_FreqL2_Main = 60.010000
10b4 : ISLAND_VL2N_Load = 121.000000
10d1 : ISLAND_FreqL2_Load = 60.010000
10f0 : ISLAND_PhaseL2_Main_Load = 0.000000
1115 : ISLAND_VL3N_Main = 0.000000
1132 : ISLAND_FreqL3_Main = 0.000000
1151 : ISLAND_VL3N_Load = 0.000000
116e : ISLAND_FreqL3_Load = 0.000000
118d : ISLAND_PhaseL3_Main_Load = -256.000000
11b2 : ISLAND_L1L2PhaseDelta = -256.000000
11d4 : ISLAND_L1L3PhaseDelta = -256.000000
11f6 : ISLAND_L2L3PhaseDelta = -256.000000
1218 : ISLAND_GridState = ISLAND_GridState_Grid_Compliant
124d : ISLAND_L1MicrogridOk = TRUE
1267 : ISLAND_L2MicrogridOk = TRUE
1281 : ISLAND_L3MicrogridOk = FALSE
129b : ISLAND_ReadyForSynchronization = TRUE
12bf : ISLAND_GridConnected = TRUE
12d9 : SYNC_ExternallyPowered = FALSE
12f5 : SYNC_SiteSwitchEnabled = TRUE
```

fkhera, 2021-08-08:
Unsubscribe

vloschiavo, 2021-08-10:
Nice work everyone!  Thanks!

Is anyone up for creating a sample decoder script?

wiedmann, 2021-09-01:
I'll just add here that putting my output through a protobuf decoder yields something at least half-way sensible, so this may be a serialization format based on a similar principle. Try this: https://protogen.marcgravell.com/decode

The strings and varints seem to decode correctly, but the fixed64 values don't appear correct, so it's probably some other format. In terms of unpacking the data, though, this might be helpful.

I'm still on version 21.20.6, FWIW.

brianhealey, 2021-09-21:
Not perfect..  but close:

`syntax = "proto3";
option go_package = "github.com/example/path/gen;gen";
package teslapower;

import "google/protobuf/timestamp.proto";

message DeviceVital {
  optional string name = 1;
  oneof value {
    uint32 intValue = 3;
    float floatValue = 4;
    string stringValue = 5;
    bool boolValue = 6;
  }
}

message ConnectionParameters {
  optional string ipAddress = 1;
  optional string serialPort = 2;
  optional string serialBaud = 3;
  optional uint32 modbusId = 4;
}

message TeslaHardwareId {
  optional uint32 pcbaId = 1;
  optional uint32 assemblyId = 2;
  optional uint32 usageId = 3;

}

message TeslaEnergyEcuAttributes {
  optional int32 ecuType = 1;
  optional TeslaHardwareId hardwareId = 2;

}

message GeneratorAttributes {
  optional float nameplateRealPowerW = 1;
  optional float nameplateApparentPowerVa = 2;
}

message PVInverterAttributes {
  optional float nameplateRealPowerW = 1;
}

message MeterAttributes {
  repeated uint32 meterLocation = 1;
}

message Device {
  optional string din = 1;
  optional string partNumber = 2;
  optional string serialNumber = 3;
  optional string manufacturer = 4;
  optional string siteLabel = 5;
  optional string componentParentDin = 6;
  optional string firmwareVersion = 7;
  optional google.protobuf.Timestamp firstCommunicationTime = 8;
  optional google.protobuf.Timestamp lastCommunicationTime = 9;
  optional ConnectionParameters connectionParameters = 10;
  oneof deviceAttributes {
    TeslaEnergyEcuAttributes teslaEnergyEcuAttributes = 11;
    GeneratorAttributes generatorAttributes = 12;
    PVInverterAttributes pvInverterAttributes = 13;
    MeterAttributes meterAttributes = 14;
  };
}

message SiteControllerConnectedDevice {
  optional Device device = 1;
}

message SiteControllerConnectedDeviceWithVitals {
    optional SiteControllerConnectedDevice device = 1;
    repeated DeviceVital vitals = 2;
    repeated string alerts = 3;
}

message DevicesWithVitals {
  repeated SiteControllerConnectedDeviceWithVitals devices = 1;
}`

brianhealey, 2021-09-21:
`syntax = "proto3";
option go_package = "github.com/example/path/gen;gen";
package teslapower;

import "google/protobuf/timestamp.proto";

message DeviceVital {
  optional string name = 1;
  oneof value {
    int64 intValue = 3;
    double floatValue = 4;
    string stringValue = 5;
    bool boolValue = 6;
  }
}

message StringValue {
  string value = 1;
}

message UInt32Value {
  uint32 value = 1;
}

message ConnectionParameters {
  optional string ipAddress = 1;
  optional string serialPort = 2;
  optional string serialBaud = 3;
  optional uint32 modbusId = 4;
}

message TeslaHardwareId {
  optional UInt32Value pcbaId = 1;
  optional UInt32Value assemblyId = 2;
  optional UInt32Value usageId = 3;

}

message TeslaEnergyEcuAttributes {
  optional int32 ecuType = 1;
  optional TeslaHardwareId hardwareId = 2;

}

message GeneratorAttributes {
  optional uint64 nameplateRealPowerW = 1;
  optional uint64 nameplateApparentPowerVa = 2;
}

message PVInverterAttributes {
  optional uint64 nameplateRealPowerW = 1;
}

message MeterAttributes {
  repeated uint32 meterLocation = 1;
}

message DeviceAttributes {
  oneof deviceAttributes {
    TeslaEnergyEcuAttributes teslaEnergyEcuAttributes = 1;
    GeneratorAttributes generatorAttributes = 2;
    PVInverterAttributes pvInverterAttributes = 3;
    MeterAttributes meterAttributes = 4;
  };
}

message Device {
  optional StringValue din = 1;
  optional StringValue partNumber = 2;
  optional StringValue serialNumber = 3;
  optional StringValue manufacturer = 4;
  optional StringValue siteLabel = 5;
  optional StringValue componentParentDin = 6;
  optional StringValue firmwareVersion = 7;
  optional google.protobuf.Timestamp firstCommunicationTime = 8;
  optional google.protobuf.Timestamp lastCommunicationTime = 9;
  optional ConnectionParameters connectionParameters = 10;
  repeated DeviceAttributes deviceAttributes = 11;
}

message SiteControllerConnectedDevice {
  optional Device device = 1;
}

message SiteControllerConnectedDeviceWithVitals {
    repeated SiteControllerConnectedDevice device = 1;
    repeated DeviceVital vitals = 2;
    repeated string alerts = 3;
}

message DevicesWithVitals {
  repeated SiteControllerConnectedDeviceWithVitals devices = 1;
}`

fkhera, 2021-09-21:
Unsubscribe

mir0810, 2021-11-07:
I logged into my TEG on its internal wifi with installer access. the "System" page shows all the vital statistics nicely represented. The reason I was in there was to figure out how much electricity each solar string was generating. But it would be nice to parse out the additional values like temp and frequency as well.
Since the website shows it clear text, I assume there is a javascript on the page that does the decoding but I haven't looked at it in detail yet :)

jasonacox, 2021-11-23:
@sean-user Thanks for your code. I ported a version to python [here](https://github.com/jasonacox/pypowerwall/blob/main/examples/vitals.py) but it isn't quite right. The `protobuf` path seems very interesting.

@brianhealey Were you able to parse with that?  Any quick tips for someone just learning protobuf? :)  I just `brew install protobuf` and checking out how to get a proto definition into python to parse these *vitals* payloads.  Going through `protoc` now...

jasonacox, 2021-11-28:
Update: I used the great work by @brianhealey (tesla.proto definition) to process the vitals payload and produce a json representation. Work in progress: https://github.com/jasonacox/pypowerwall/tree/main/examples/vitals

reli3, 2021-12-05:
What would be the best way to format the output a bit better?

I am Pre-PTO I think there are issues with my MCI - Maybe once a week / day etc. I have to keep an eye on my strings.
Sometimes 1 or maybe 2 strings are not functioning so I have to do a reboot of the inverters to make them show up again. 

If getting the fromat correct and comparing it from a baseline everyday if something turns False for a string some type of email notifaction would be amazing for a lot of people I believe when they think there solar is under producing but in the end it could be a MCI problem or a string issue since people complain alot about under performance this could be the reason why! 

After executing this is what I get for the Dispaly String Data:

{'A': {'Current': 1.1300000000000001, 'Voltage': 141.20000000000002, 'Power': 173.0, 'State': 'PV_Active', 'Connected': True}, 'B': {'Current': 0.54, 'Voltage': 339.0, 'Power': 178.0, 'State': 'PV_Active', 'Connected': True}, 'C': {'Current': 0.59, 'Voltage': 324.6, 'Power': 197.0, 'State': 'PV_Active', 'Connected': True}, 'D': {'Current': 0.48, 'Voltage': 99.4, 'Power': 64.0, 'State': 'PV_Active', 'Connected': True}, 'A1': {'Current': 1.1, 'Voltage': 302.90000000000003, 'Power': 327.0, 'State': 'PV_Active', 'Connected': True}, 'B1': {'Current': 1.09, 'Voltage': 303.5, 'Power': 320.0, 'State': 'PV_Active_Parallel', 'Connected': True}, 'C1': {'Current': 0.0, 'Voltage': -1.299999999999999, 'Power': 0.0, 'State': 'PV_Active', 'Connected': False}, 'D1': {'Current': 0.0, 'Voltage': -0.29999999999999893, 'Power': 0.0, 'State': 'PV_Active', 'Connected': False}}

jasonacox, 2021-12-05:
Hi @reli3 - I want to keep an eye on my strings as well, which is why I wrote this.  I also wanted to keep an eye on the Neruio solar meter which is also in the `vitals` payload.   For string data, if you want to see it in human readable format, just specify jsonformat as True like this:
```python
strings = pw.strings(jsonformat=True)
print(strings)
```
Output:
```json
{
    "A": {
        "Connected": true,
        "Current": 0.48,
        "Power": 115.0,
        "State": "PV_Active",
        "Voltage": 251.8
    },
    "B": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -2.3999999999999995
    },
    "C": {
        "Connected": true,
        "Current": 1.82,
        "Power": 526.0,
        "State": "PV_Active",
        "Voltage": 293.6
    },
    "D": {
        "Connected": true,
        "Current": 1.81,
        "Power": 521.0,
        "State": "PV_Active_Parallel",
        "Voltage": 294.0
    }
}
```
I push that into a tiny installation of Splunk (see [TinySplunk](https://github.com/jasonacox/TinySplunk)) and get a graph like this:

[image]

I have a lot of trouble with the Neruio solar meter that Tesla installed in my Powerwall+ so I also have scripts to watch that.  It shows up in the `vitals` payload as a device that I monitor and capture to create alerts (e.g. if it doesn't show up in the `vitals` payload that means it has lost connection with the gateway and shut down solar production):

```json
"NEURIO--VAHxxxxxxxxxx": {
        "NEURIO_CT0_InstRealPower": 1333.5606370817025,
        "NEURIO_CT0_Location": "solar",
        "Parent": "STSTSM--xxxxxxx-00-E--TGxxxxxxxxxxxx",
        "firmwareVersion": "1.6.1-Tesla",
        "lastCommunicationTime": "1638720520",
        "manufacturer": "NEURIO",
        "partNumber": "",
        "serialNumber": "VAHxxxxxxxxxx"
    }
```

reli3, 2021-12-05:
@jasonacox omg thata is amazing exactly what I think I am looking for....any chance for a write up to set this up via tinysplunk on how you are getting the data in? Does it pull the string data every minute? 

This output will help in the morning and ill run it via cron to put to what I need instead of having to login via installer and flipping the powerwall switch when I think its not producing the correct amount of solar as expected! 

I loved your other dashboard with grafana was thinking of making a PI display just for that eventually.  

I believe I have the NEURIO as well since I have Two Powerwall+ and 4x Powerwalls. I haven't had any issues with the Neurio at least so far besides sometimes the inveter likes to discount from the Wi-Fi every once in a while which what I found was to re join it via the App surpisingly and it fixes it. 

Thanks for your amazing work so far with this!

jasonacox, 2021-12-05:
Thanks @reli3 !  I cross posted this as an issue under pypowerwall for Strings Data- (see https://github.com/jasonacox/pypowerwall/issues/1) since I don't know how relevant it would be for this issue.  I included the script I use to gather and push the string data into Splunk.  I plan to add this to the Grafana dashboard as well. :)

brianhealey, 2022-01-16:
Glad to see that the protocol buffer example helped. :) There were some errors in what I put together, so I forked your branch to update it.  Hopefully this update will help flush out any issues your might be seeing and/or include fields I didn't have before that might be silently ignored.

closed 2023-04-06 by vloschiavo