vloschiavo/powerwall2 #55: Possibility of a New Authentication Method?

rstad944, 2021-09-14:
I noticed in the latest Tesla Android app under Settings that there is an option to "Pair Phone to Powerwall" which requires a toggle of the PW switch.

Has anyone been able to exploit this option to pair a local controller (e.g. Raspberry Pi) to the TEG via a LAN connection?

I am especially wondering if the Powerwall mode (backup, self powered) could once again be controlled locally without using the owner-api.teslamotors.com endpoint.  This local control capability had been essentially removed when the PW firmware was updated from 1.49 to 1.50.

dan-simpson-zip, 2023-03-21:
Also keen to hear more about this!  Just tried implementing a POST to `/api/operation` in tesla_powerwall python library repository, and it wasn't till I ran an integration test that I happened upon an access denied error, e.g.:

```
tesla_powerwall.error.AccessDeniedError: Access denied for resource /api/operation: Unable to POST to resource: User does not have adequate access rights
```

Would be fantastic to modify backup percentage via api 🙏

daniel-simpson, 2023-03-21:
Also keen to hear more about this!  Just tried implementing a POST to `/api/operation` in tesla_powerwall python library repository, and it wasn't till I ran an integration test that I happened upon an access denied error, e.g.:

```
tesla_powerwall.error.AccessDeniedError: Access denied for resource /api/operation: Unable to POST to resource: User does not have adequate access rights
```

Would be fantastic to modify backup percentage via api 🙏

vloschiavo, 2023-04-06:
Agree. This does sound very interesting.  I'll leave this issue open until we can find a solution.

patniemeyer, 2024-01-02:
I was able to make API calls to my gateway by grabbing the bearer token after logging in to the local web app (as "customer").  It only lasts for a few hours but it allows me to make calls to the /vitals endpoint.

longzheng, 2024-01-26:
> I was able to make API calls to my gateway by grabbing the bearer token after logging in to the local web app (as "customer"). It only lasts for a few hours but it allows me to make calls to the /vitals endpoint.

@patniemeyer Just double checking if you're still able to replicate this? I tested using a bearer token and am still seeing a 404 for `/api/devices/vitals`.

```
GET /api/devices/vitals HTTP/1.1
Authorization: Bearer REDACTED
User-Agent: PostmanRuntime/7.36.1
Accept: */*
Host: 192.168.1.68
Accept-Encoding: gzip, deflate, br
Connection: keep-alive
 
HTTP/1.1 404 Not Found
Content-Type: text/plain; charset=utf-8
X-Content-Type-Options: nosniff
Date: Tue, 23 Jan 2024 07:37:15 GMT
Content-Length: 19
 
404 page not found
```

Sn0w3y, 2024-02-05:
@longzheng as we can see it is REDACTED.

Maybe we can implement a Authentication to the Code?

Greetings

Matthew1471, 2025-10-30:
I'm working on this.. it involves a lot of protobuffs and generating encryption/decryption keys. The power toggle is to "accept" a public encryption key.

This is in line with the vehicles too, so some more info is [here](https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md) and [here](https://github.com/teslamotors/vehicle-command/blob/main/README.md)