vloschiavo/powerwall2 #57: sitemaster/stop

gemu2015, 2021-09-29:
i want to stop my powerwall when i charge my electric car because its silly to charge the car battery with the house battery.
unfortunately   sitemaster/stop  returns "method not allowed"
do you have advice how to achieve that command ?

sitemaster/run executes successfully

BJReplay, 2021-09-29:
You may be able to do that if logged in as customer.

However, setting the backup reserve percentage to the current state of
charge has the same effect - puts the powerwall into standby so that it
doesn't discharge into load (whether that is your car or household load).

foogod, 2021-12-26:
I believe the problem is that I've determined the `sitemaster/stop` API call is not actually a GET request as indicated here (this may have changed in more recent versions of the API).  It needs to be a POST with the following body:

```
{"force":false}
```

(If `force` is set to `false`, then it will only stop sitemaster if it is idle (i.e. it is not supplying a significant amount of power to the house, and is not in the middle of something like a firmware download/update, etc), which is probably what you want to do for automated things.  If `force` is set to `true` it will stop it even if doing so would interrupt something it's in the middle of, or interrupt supplying power to the house, etc (`{"force":true}` is sent by the web interface when you actually click on the "Stop System" button, for example))

foogod, 2021-12-26:
Note also that if you actually stop the sitemaster, then the powerwall will not do *anything*.  That means it also can't charge the battery, or provide backup power in the case of a power outage, etc.  Because of this, something like BJReplay suggested of just adjusting the reserve percentage may be a better approach anyway..

gemu2015, 2021-12-27:
@foogod 
thanks for your explanation, i will try that.

off topic:
i would like to ask if other Powerwall users have the same problem:
i have had several incidents of total discharge of the powerwall to zero percent.
this happens when there is nearly no solar  input on very dark winter times.
this is the 3. year that this happens on dark days. i contacted Tesla several times and they denied this would be a firmware problem. they simply increased the reserve percentage to 8 percent what did not help at all.
i am heavily concerned about this behavior that it might destroy my battery.
the majority of users do not monitor the power walls local api and will never recognize this problem.
what do you think ?

[image]

emanuelbalea, 2021-12-27:
Yes, there is a max site import setting. If its set and you import power more than that then it will use the battery regardless of its reserve setting. This was auto defaulted for me during an update to 5kw when it should have been 0 as i dont need to limit import.