vloschiavo/powerwall2 #73: Getting a certificate on windows

kemble9900, 2022-05-21:
Hi, I just wanted to ask how to go about getting a certificate on windows. I tried using 

openssl s_client -showcerts -connect 192.168.0.164:443 < /dev/null | sed --quiet '/BEGIN CERTIFICATE/,/END CERTIFICATE/p' > cacert.pem

however it said the system cannot find the path specified, so I wondered if anyone knew where I could be going wrong,
thanks, kemble9900.

gmichaela, 2022-11-01:
openssl s_client -showcerts -connect 192.168.0.164:443

and copy and paste the 
-----BEGIN CERTIFICATE-----
to
-----END CERTIFICATE-----
block.

(or output to a file by adding to the end > c:\temp\cacert.pem)  and keep the above block after editing.

gmichaela, 2022-11-04:
Even easier on windows--
Fire up Chrome, where the HTTPS lock is typically ('Not Secure') click,  Certificate is not valid, click.   Details tab.  Export... button

closed 2023-04-06 by vloschiavo